Xinyuan-LilyGO/LilyGo-Modem-Series #4: missing deep sleep current in documentation

ClemensGruber, 2021-11-18:
I'm following the most publications of new GSM / LTE form LilyGo and waiting still for a really low power board in deep sleep (< 20 uA). The combination of (optinal) Wifi, GSM / LTE with solar charging IC is a often wished use case and you can see less boards on this sector. So I'm happy that LilyGo develops a bunch of boards here. The only thin I'm missing here is really a low deep sleep power consumption. 

Are there numbers for the A7670x board in deep sleep? LilyGo's T-Call SIM800 has arround 500 uA and the T-SIM SIM7000G 250 uA as far as I remember.

ClemensGruber, 2021-12-06:
Btw. nice extension that VBat is now a dedicated pin. With the JST plug only (like the T-Call SIM800) as LiPo input some scenarios are not good feasible. E.g. we had this scenario tried out with a timer and solar charger in the power line but it does not work good with a cable connection: https://community.hiveeyes.org/uploads/default/original/2X/b/bcfe0dbe759cdd30cbfbd222692575fe54646850.png

Was the LilyGO-T-A7670X just a trial, I did not find it on aliexpress anymore or will it be available in the future. Would be still interested in the deep sleep values.

LilyGO, 2022-04-18:
The T-A7670 has a dormancy of around 170uA

closed 2022-04-18 by LilyGO