Xinyuan-LilyGO/LilyGo-Modem-Series #8: Problem about gps/gnss

emrecanozkok, 2022-04-08:
I bought this board and did some testing. The sd card and gprs feature are working but all my other devices are showing they can't get gps signal where they are working. I'm just running the demo code and I keep getting a blank gps signal message. I need help on this.

[image]

LilyGO, 2022-04-09:
Hello. Did you buy the board with GPS support?

emrecanozkok, 2022-04-09:
> Hello. Did you buy the board with GPS support?

I heated the gps antenna connection and then it started working, but it is more difficult to detect than other modules. Which should I use before starting hot,warm or cold.Is there an adc pin on this module where I can read the battery voltage? and is there a led i can control. I couldn't find a detailed explanation.

LilyGO, 2022-04-12:
GPIO35可以获取电池的电压

emrecanozkok, 2022-04-12:
yes you are right im reading voltage when usb plugged it is true but when i disconnect device and run with battery its not read properly its getting everytime 4.2 volt

LilyGO, 2022-04-18:
float readBattery(uint8_t pin)
{
    int vref = 1100;
    uint16_t volt = analogRead(pin);
    float battery_voltage = ((float)volt / 4095.0) * 2.0 * 3.3 * (vref);
    return battery_voltage;
}

You can refer to this function.  There will be some error because of the partial pressure.  And you get a little bit of an error

closed 2022-04-30 by emrecanozkok