Xinyuan-LilyGO/LilyGo-Modem-Series #10: New version has GPIO5 connected to RST of A7670E

MilkoKrastev, 2022-05-10:
I have a nasty problem with A7670E resetting each time ESP32 WROVER-E exits deep sleep. My program flow is the following:

1. A7670 is in sleep mode (AT+CSCLK=1, DTR HIGH) and not powered down. 
2. ESP32 is in deep sleep while the ULP is active collecting some sensor data. 
3. The ULP wakes up the main cores and ESP32 tries to publish some data via A7670. 
4. However, each time ESP32 exits deep sleep, GPIO5 is pulled up (it is strapping pin) and A7670 is reset (restarted). 

Is there a way to avoid that? The previous version of this board did not have GPIO5 connected to A7670's RST.

LilyGO, 2022-05-11:
A7670 reset.  
Do you have these lines of code in your program?  Comment it out?  

 
[image]

MilkoKrastev, 2022-05-11:
> Do you have these lines of code in your program? annotation it.

The module is reset with or without these lines of code. I think this may have to do with the fact that the RESET pin of A7670 is connected to GPIO 5 of ESP32 which is pulled up at boot. Then GPIO 5 goes to 0, and this is like a reset. 

Is there a way to remove connection between GPIO 5 and the RESET pin of A7670? Do you have a sample code where ESP32 goes from deep sleep to normal mode without resetting A7670 (A7670 is ON while ESP32 is in deep sleep)? This setup used to work great with the previous version of this board where GPIO 5 was not connected to the RESET pin of A7670.

LilyGO, 2022-05-13:
GPIO12 is now the control power pin, which has a role at low power consumption.  
To use GPIO5 you can remove the resistor as shown.  
If you have more custom requirements, you can customize the board to meet your requirements. 
[image]

MilkoKrastev, 2022-05-13:
> To use GPIO5 you can remove the resistor as shown.
If you have more custom requirements, you can customize the board to meet your requirements.

Thanks a lot for that info! However, I have several boards which are version 1.0 (2021-12-18) and not 1.1 (2022-2-28). Could you please confirm that the location of the resistor for GPIO 5 is the same for 1.0 as for 1.1? Thanks