Xinyuan-LilyGO/LilyGo-Modem-Series #11: Upload of Sketch to ESP32 not working

mjagemann, 2022-06-23:
Hi there,

I do have a A7670E and try to upload a sketch to the ESP32 on Win10  using Arduino.
I've installed the CH9102 which is available from Lilygo here on Github, how ever the Board does not react to uploads.

What to do? Do I have to put the board into upload/flash mode somehow?

LilyGO, 2022-06-24:
Do you have com ports on your computer?  Does your data line support burning?

mjagemann, 2022-06-24:
Yes, when I plug in the board, 4 com ports show up. Modem is also recognized.
2 of these ports respond to AT commands succesfully. 1 of these ports show debug information continously.

Trying to upload a sketch on port 1-3 gives following feedback:
`esptool.py v3.0-dev
Serial port COM6
Connecting........_____....._____....._____....._____....._____....._____.....____Beim Hochladen des Sketches ist ein Fehler aufgetreten
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
`

Trying on the port which prints debug gives following:
`esptool.py v3.0-dev
Serial port COM9
Connecting........_____....._____....._____....._____....._____....._____.....____Beim Hochladen des Sketches ist ein Fehler aufgetreten
_

A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x16)
`
Not sure what you mean with "Does your data line support burning?".
I am able to upload sketches to other ESP32s with the same computer and cable.

LilyGO, 2022-06-24:
You need type-c. Is the serial port of ESP32

mjagemann, 2022-06-24:
OK, usb-c powers the board but does not show a com port (also not an unrecognized device).

I installed this driver, is this the right one?
https://github.com/Xinyuan-LilyGO/CH9102_Driver

(P.S. a little off topic but the community board of LilyGo does not allow new registrations. It requires email confirmation but no email will be sent. Checking the MAilsettings in the profile gives following error. 

Please review the entered information:
Invalid login: 535 Error: authentication failed, system busy
)

LilyGO, 2022-06-24:
[Try downloading this]( https://www.wch.cn/downloads/CH343SER_EXE.html)

mjagemann, 2022-06-24:
Found out it was the cable, never thought about that. 
New cable and upload works!!!

Thanks for your support

closed 2022-06-24 by mjagemann