Xinyuan-LilyGO/LilyGo-Modem-Series #13: Problem with reading battery voltage

MilkoKrastev, 2022-07-03:
Reading the battery voltage using ADC Channel 7/GPIO35 does not work as expected. The value read is always the same although the battery voltage level drops. The module is not powered via USB, it runs only on battery. 

Could you please post ESP IDF sample code that demonstrates how the battery voltage should be retrieved? My code works fine with TTGO SIM7000G, but does not work with A7670E. Any idea what differences between TTGO SIM7000G and A7670E could be causing that?

Thanks!

LilyGO, 2022-07-04:
Hello.
Refer to this [sketch](https://github.com/Xinyuan-LilyGO/LilyGO-T-A7670X/blob/main/examples/new_version/Arduino/battery_voltage/battery_voltage.ino)

MilkoKrastev, 2022-07-04:
Hi, I uploaded the Arduino sketch to one of my boards, and it shows

0.28V

which is obviously wrong. Any idea what could be wrong? I have not changed the example code in any way.

LilyGO, 2022-07-05:
The sketch has been optimized.  Which version of the T-A7670 do you use?  What is the silkscreen?

MilkoKrastev, 2022-07-05:
I test with two boards getting the same result. One of the boards is version 1.0, the other one version 1.1.

MilkoKrastev, 2022-07-05:
I tried the updated sketch for battery voltage, but it always prints 3.88V regardless of the actual voltage in the battery. I tried with two different batteries with different voltage levels none of which equals 3.88V. 

Any other ideas what to try next?

LilyGO, 2022-07-06:
TTGO SIM7000G Detects the battery voltage.  The A7670 version does not currently have this feature.  This feature will be added in future optimizations .

closed 2022-07-06 by LilyGO