Xinyuan-LilyGO/LilyGo-Modem-Series #15: No GNSS data

gnistregn, 2022-07-21:
Saw a previous ticket about this issue, but the "resolution" there didn't make any sense. Anyhow, I have a hard time getting the GNSS data to be anything other than just:
```

AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
getGPS 86981
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
getGPS 88988
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK

```

Tried two different antennas, tested the connector on the PCB if it was damaged somehow, but nothing seems to work.
I have the unit in my window, and have let it run for more than 30 minutes without a change.
My unit version is "2022-5-22 V1.1 XY--A7670"

LilyGO, 2022-07-22:
Hello，Are you using this [example?](https://github.com/Xinyuan-LilyGO/T-A7670X/tree/main/examples/new_version/Arduino/TinyGSM_Net_GNSS) Did you buy the version with GNSS?

gnistregn, 2022-07-22:
Yes, that's the example I'm using.
I bought the version with GNSS (from Banggood). GNSS antenna was included also, which I took as confirmation that I received the right type of unit.

LilyGO, 2022-07-25:
Is it positioned outdoors?

rustyw, 2022-07-28:
> 

Hi gnistregn,
 
I am having similar issues, what are the pre-commands you sending before enabling the GPS as on previous versions you had to enable the aerial booster voltage pin for it to function.

Regards
Russ

closed 2022-08-18 by LilyGO