Xinyuan-LilyGO/LilyGo-Modem-Series #16: SD card not functioning

rustyw, 2022-07-27:
Using the supplied firmware that came with the board and trying different SD cards using FAT, FAT32 I am not able to get it to work.
Even when trying SD firmware that works on the LilyGo SIM7000G I get the same results. I have double checked the CS, MISO, MOSI and CLK wires but to no avail. There are definitely issues with these boards.

LilyGO, 2022-07-28:
Hello. These boards are tested with SD cards in factory testing. Did you buy a T-A7670 or a T-SIM7000? Sometimes it's a bug in the esp32 package. What is your current esp32 version? It should have been fixed in the latest package, please refer to the changelog of the esp32 package!

rustyw, 2022-07-28:
He

Here are the images of the new board.

[image]
[image]

Thank you.

LilyGO, 2022-07-29:
Running this [sketch](https://github.com/Xinyuan-LilyGO/T-A7670X/tree/main/examples/new_version/Arduino/TinyGSM_Net_GNSS) can detect TF cards?

closed 2022-08-18 by LilyGO