Xinyuan-LilyGO/LilyGo-Modem-Series #18: Deep sleep mode

BJ4K3, 2022-08-02:
Hi, I would like to put the board in deep sleep, I've tried turning off the modem with the `AT+CPOF` AT command (naturally I've set the esp32 in hibernation mode) but the total consumption is around 66mA and not the 0.18mA you indicated. Can I have some sort of code to put the entire board in sleep mode reaching the current consumption you mentioned?

Thank you.

LilyGO, 2022-08-03:
Reference [sketch](https://github.com/Xinyuan-LilyGO/T-A7670X/blob/main/examples/new_version/Arduino/TinyGSM_Net_GNSS/TinyGSM_Net_GNSS.ino). sleep at the end.

BJ4K3, 2022-08-05:
Hi, as said I have already tried the AT+CPOF command (like the tinygsm library) but the board absorb too much current even after the command and the ESP32 in hibernate mode.

LilyGO, 2022-08-08:
Did you run the sketch above?

LilyGO, 2022-08-18:
No reply for a long time. Close

closed 2022-08-18 by LilyGO