Xinyuan-LilyGO/LilyGo-Modem-Series #19: measured battery voltage does not change

nukutoki, 2022-08-08:
Hi,
I am using the battery voltage [example](https://github.com/Xinyuan-LilyGO/T-A7670X/blob/main/examples/new_version/Arduino/battery_voltage/battery_voltage.ino).
The measured battery voltage is always 4.04V (if no USB is connected). I am using [this](https://pt.aliexpress.com/item/1005003036514769.html) board. Should the example work on this board or do I have to connect the GPIO35 somehow?
Thanks in advance.

LilyGO, 2022-08-09:
Hello.The battery check function is available only for 2022-7-28 V1.2.

closed 2022-08-09 by LilyGO

nukutoki, 2022-08-09:
OK, thank you for the fast reply.

giannoug, 2022-09-28:
@LilyGO So how can we check battery voltage for 2022-5-22 V1.1?

LilyGO, 2022-09-29:
Hello. Only V1.2 has the function of detecting battery voltage

giannoug, 2022-09-29:
Which pin of the board has battery voltage? VBAT? We could be able to use a voltage divider to calculate it

LilyGO, 2022-09-29:
V1.2.    The IO35 can capture battery voltage

LilyGO, 2022-09-29:
I'm sorry. V1.1 does not have a GPIO for ADC to detect battery voltage