Xinyuan-LilyGO/LilyGo-Modem-Series #20: GPS Example

fcojaviergl, 2022-09-24:
Hi, 

I am new with Arduino, I have the board A7670 v1.2, and I would like to test the GPS, but I can't use internet examples, can you upload a GPS example in the repo?

Thanks.

LilyGO, 2022-09-24:
Hello. You can use this [example](https://github.com/Xinyuan-LilyGO/T-A7670X/tree/main/examples/new_version/Arduino/TinyGSM_Net_GNSS)

fcojaviergl, 2022-09-24:
Thanks, I didn't see it.

closed 2022-09-24 by fcojaviergl

walidbouachour, 2023-04-14:
hi , so i have a error in this example ,can you help me ?

gps:285:5: error: 'SerialMon' was not declared in this scope

     SerialMon.print("\tWait GPS reday.");

     ^

Plusieurs bibliothèque trouvées pour "SD.h"

exit status 1

'SerialMon' was not declared in this scope