Xinyuan-LilyGO/LilyGo-Modem-Series #21: How does BAT_EN pin work?

giannoug, 2022-10-05:
I can't seem to understand how BAT_EN (IO12) works. I have a board versioned 2022-5-22 V1.1. Can you explain?

Both do nothing at all, battery is controlled only with switch:
```
  pinMode(BAT_EN, OUTPUT);
  digitalWrite(BAT_EN, LOW);
```
same for
```
  pinMode(BAT_EN, OUTPUT);
  digitalWrite(BAT_EN, HIGH);
```

LilyGO, 2022-10-06:
Hello.  When only battery power is available. You need to enable BAT_EN for power supply

giannoug, 2022-10-06:
On battery power, battery is only controlled with the switch, `digitalWrite(BAT_EN, LOW);` and `digitalWrite(BAT_EN, HIGH);` do nothing 🤔

LilyGO, 2022-10-27:
digitalWrite(BAT_EN, LOW);   Unable to start A7670

giannoug, 2022-10-28:
>digitalWrite(BAT_EN, LOW); Unable to start A7670

Doesn't make sense according to this:
[image]

LilyGO, 2022-10-29:
This circuit is in good working order. Can you elaborate on why it doesn't make sense?

giannoug, 2022-10-29:
When the board runs on battery power, the power can be cutoff using the BAT_EN (IO12) pin if I understand correctly. It doesn't work in my case

gasagna, 2022-10-29:
> digitalWrite(BAT_EN, LOW); Unable to start A7670

This is correct, as I demonstrate [here](https://github.com/Xinyuan-LilyGO/T-A7608X/issues/8) for a slightly different dev board and a slightly different problem. Basically, if `BAT_EN` remains low, the simcom module cannot be woken up while on battery power, e.g. with a pulse on the `POWERKEY`. While attached to the USB, this pin does nothing. 

The nice thing is that if you keep the `BAT_EN` pin low, the simcom module never wakes up automatically. This is nice in low-power applications, when you might only want to turn on the modem every hour of so to send data to a server.

These are great boards with lots of functionality, but I wish the documentation was more extensive

gasagna, 2022-10-29:
> When the board runs on battery power, the power can be cutoff using the BAT_EN (IO12) pin if I understand correctly. It doesn't work in my case

I do not think that is the intended behaviour (at least on my board with a A7608). The ESP32 keeps working, regardless of `BAT_EN`. It's the simcom module that depends on it.

giannoug, 2022-10-29:
@gasagna I agree they are good boards, but documentation is all over the place.

For example check this: https://github.com/Xinyuan-LilyGO/T-A7670X/blob/main/examples/new_version/Arduino/battery_voltage/battery_voltage.ino#L17

Why does this sketch "enable battery" when it's not going to use the A7608?

Also check the pinout on the homepage:
[image]
It clearly shows that the 18650 is enabled with pin 12.

Also, there is a battery voltage sketch for V1.0 boards but [that functionality is missing from these boards](https://github.com/Xinyuan-LilyGO/T-A7670X/issues/19#issuecomment-1261632135)... https://github.com/Xinyuan-LilyGO/T-A7670X/blob/main/examples/new_version/Arduino/battery_voltage/battery_voltage.ino

@LilyGO Lastly let's not forget this product costs ~$50. Two of the "advertised" features do not work.

LilyGO, 2022-10-31:
From the schematic, we know that BAT_EN(IO12) controls whether BAT is on or off. When using only battery power VBUS is not. BAT will supply power to A7670. So BAT_EN(IO12) is the on/off of BAT to control the power of A7670.

closed 2022-11-14 by LilyGO

daveymg-nz, 2022-11-17:
> > digitalWrite(BAT_EN, LOW); Unable to start A7670
> 
> Doesn't make sense according to this: [image]

Hi, sorry to hijack the thread, but could I ask where you obtained the schematic from?