Xinyuan-LilyGO/LilyGo-Modem-Series #25: A7670SA R2 With GPS  not works

felipesms, 2022-11-12:
Hello , So I have the module T-SIM-A7670-R2 (A7670SA R2 With GPS)  I am using Arduino to program.
I able to connect to the network but I couldn't get the GPS to work.

AT+CGNSSINFO
+CGNSSINFO: ,,,,,,,,

thank you so much

LilyGO, 2022-11-12:
Hello. Which example are you using? Also make sure the antenna is outdoors

felipesms, 2022-11-12:
thanks...
I tested outdoors with antenna

my code:
------------------------------------------

#include "Arduino.h"

#define TINY_GSM_MODEM_SIM7600  //The AT instruction of A7670 is compatible with SIM7600 
#define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb
#define SerialAT Serial1

// See all AT commands, if wanted
#define DUMP_AT_COMMANDS

#define uS_TO_S_FACTOR 1000000ULL  // Conversion factor for micro seconds to seconds
#define TIME_TO_SLEEP  600          // Time ESP32 will go to sleep (in seconds)

#define UART_BAUD    115200
#define PIN_DTR      25
#define PIN_TX       26
#define PIN_RX       27
#define PWR_PIN      4
#define BAT_ADC      35
#define BAT_EN       12
#define PIN_RI       33
#define RESET        5

#define SD_MISO     2
#define SD_MOSI     15
#define SD_SCLK     14
#define SD_CS       13
// Your GPRS credentials, if any
const char apn[]  = "zap.vivo.com.br";
const char gprsUser[] = "vivo";
const char gprsPass[] = "vivo";

#include "int.cpp"

#include <TinyGsmClient.h>

#ifdef DUMP_AT_COMMANDS  // if enabled it requires the streamDebugger lib
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, Serial);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

void modem_on()
{

    pinMode(BAT_EN, OUTPUT);
    digitalWrite(BAT_EN, HIGH);

    //A7670 Reset
    pinMode(RESET, OUTPUT);
    digitalWrite(RESET, LOW);
    delay(100);
    digitalWrite(RESET, HIGH);
    delay(3000);
    digitalWrite(RESET, LOW);

    pinMode(PWR_PIN, OUTPUT);
    digitalWrite(PWR_PIN, LOW);
    delay(100);
    digitalWrite(PWR_PIN, HIGH);
    delay(1000);
    digitalWrite(PWR_PIN, LOW);

}

bool reply = true;

// liga gps e flash 
// AT+CGNSSPWR=1,1 

//AT+CGPSINFO
void setup()
{
    Serial.begin(115200); // Set console baud rate
    SerialAT.begin(115200, SERIAL_8N1, PIN_RX, PIN_TX);
    delay(10);
    modem_on();

    modem.gprsConnect(apn, gprsUser, gprsPass);
    delay(2000);

    ****//turn on gps
    Serial.println("\n---Starting GPS TEST---\n");
    SerialAT.println("AT+CGNSSPWR=1,1");****  
    
    delay(10000);

    

    if (modem.isGprsConnected()) {
        Serial.println("GPRS connected");
    }

    String ccid = modem.getSimCCID();
    Serial.println("CCID: " + ccid);

    String imei = modem.getIMEI();
    Serial.println("IMEI: " + imei);

    String cop = modem.getOperator();
    Serial.println("Operator: " + cop);

    IPAddress local = modem.localIP();
    Serial.println("Local IP: " + String(local));

    int csq = modem.getSignalQuality();
    Serial.println("Signal quality: " + String(csq));

    SerialAT.println("AT+CPSI?");     //Get connection type and band
    modem.getModemInfo();

 // get gps data

    float lat,  lon;
    if (modem.getGPS(&lat, &lon)) {     Serial.printf("lat:%f lon:%f\n", lat, lon);}

    if (reply) {
        Serial.println(F("***********************************************************"));
        Serial.println(F(" You can now send AT commands"));
        Serial.println(F(" Enter \"AT\" (without quotes), and you should see \"OK\""));
        Serial.println(F(" If it doesn't work, select \"Both NL & CR\" in Serial Monitor"));
        Serial.println(F(" DISCLAIMER: Entering AT commands without knowing what they do"));
        Serial.println(F(" can have undesired consiquinces..."));
        Serial.println(F("***********************************************************\n"));
    } else {
        Serial.println(F("***********************************************************"));
        Serial.println(F(" Failed to connect to the modem! Check the baud and try again."));
        Serial.println(F("***********************************************************\n"));
    }
}

void loop()
{

    while (true) {
        if (SerialAT.available()) {
            Serial.write(SerialAT.read());
        }
        if (Serial.available()) {
            SerialAT.write(Serial.read());
        }
        delay(10);
    }

}

LilyGO, 2022-11-14:
[Reference example](https://github.com/Xinyuan-LilyGO/T-A7670X/blob/main/examples/new_version/Arduino/TinyGSM_Net_GNSS/TinyGSM_Net_GNSS.ino#L271)

felipesms, 2022-11-14:
Thanks, I used the code and add data APN. I sended for board. the chip not returned message of gps . Best regards for solutions.
thanks. I from brazil, need  the chip  any command?

// log port com

getGPS 463441
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
getGPS 465446
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
getGPS 467451
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
getGPS 469456
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
getGPS 471461
AT+CGNSSINFO

+CGNSSINFO: ,

felipesms, 2022-11-14:
A7670SA-FASE

OK
Modem Name: A7670SA-FASE

Manufacturer: INCORPORATED
Model: A7670SA-FASE
Revision: A7670M7_V1.11.1
IMEI: 869731052065927
+GCAP: +CGSM,+FCLASS,+DS

LilyGO, 2022-11-15:
Is the antenna wired right. Can you take a picture of your board?

felipesms, 2022-11-15:
link
https://imgur.com/a/jLZ0BI8

LilyGO, 2022-11-15:
Leave the example unchanged. Connect the MicroUSB and open the serial port with NMEA. See if there is data output . (The board needs to be positioned outdoors)

felipesms, 2022-11-15:
UHHHHHHHHHHHHHHH

CGNSSINFO: ,,,,,,,,

OK
getGPS 59215
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
getGPS 61220
AT+CGNSSINFO

+CGNSSINFO: 3,08,,00,00,1947.80135,S,04354.86991,W,151122,124230.00,802.6,0.414,,12.,5.1,10.

OK
lat:-0.000000 lon:-0.000000
AT+CGPS=0

All ready.

thanks

felipesms, 2022-12-12:
Hello,
thank you very much.
I will try .

bluebelts, 2022-12-17:
LilyGO-T-SIM7000G works Great! This one A7670SA R2 With GPS do not get GPS coordinates, no matter what we try, all the commands above.  Opening dispute on store to send it back.

lewisxhe, 2022-12-28:
> Hello, thank you very much. I will try .

Please try again according to the above command

i-e-b, 2023-02-07:
Hi. I have the T-SIM T-L7670E

I have uploaded the AT Debug, and gone through the process above.
I get the diagnostic information, but `AT+CGNSSINFO` and `AT+CGPSINFO` still always return blank.

What am I doing wrong, please?

```
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
$GNRMC,172922.00,V,,,,,,,070223,,,N,V*10
$GNGGA,172922.00,,,,,0,00,99.99,,,,,,*75
$GNGLL,,,,,172922.00,V,N*59
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,03,05,,,31,16,,,24,02,,,18,0*6B
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
C$GNRMC,172922.50,V,,,,,,,070223,,,N,V*15
$GNGGA,172922.50,,,,,0,00,99.99,,,,,,*70
$GNGLL,,,,,172922.50,V,N*5C
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,03,05,,,30,16,,,24,02,,,17,0*65
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
$GNRMC,172923.00,V,,,,,,,070223,,,N,V*11
$GNGGA,172923.00,,,,,0,00,99.99,,,,,,*74
$GNGLL,,,,,172923.00,V,N*58
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,03,05,,,30,16,,,25,02,,,17,0*64
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
GN$GNRMC,172923.50,V,,,,,,,070223,,,N,V*14
$GNGGA,172923.50,,,,,0,00,99.99,,,,,,*71
$GNGLL,,,,,172923.50,V,N*5D
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,03,05,,,29,16,,,20,02,,,17,0*69
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
S$GNRMC,172924.00,V,,,,,,,070223,,,N,V*16
$GNGGA,172924.00,,,,,0,00,99.99,,,,,,*73
$GNGLL,,,,,172924.00,V,N*5F
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.9S9,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,03,05,,,29,16,,,20,02,,,17,0*69
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
$GNRMC,172924.50,V,,,,,,,070223,,,N,V*13
$GNGGA,172924.50,,,,,0,00,99.99,,,,,,*76
$GNGLL,,,,,172924.50,V,N*5A
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,03,05,,,28,16,,,20,02,,,17,0*68
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
[... 94 lines cut ...]
$GPGSV,1,1,03,05,,,29,16,,,21,02,,,15,0*6A
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
,0$GNRMC,172930.00,V,,,,,,,070223,,,N,V*13
$GNGGA,172930.00,,,,,0,00,99.99,,,,,,*76
$GNGLL,,,,,172930.00,V,N*5A
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,03,05,,,29,16,,,21,02,,,15,0*6A
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E

OK
AT+CGPSINFO
+CGPSINFO: ,,,,,,,,

OK
AT+CGNSSINFO
+CGNSSINFO: ,,,,,,,,

OK
```

closed 2023-02-11 by lewisxhe

lewisxhe, 2023-04-27:
It can be seen that you did not operate according to the above method. When you turn on the GPS, you must turn on the power supply of the end of the world, otherwise you will not receive any satellite signals

https://github.com/Xinyuan-LilyGO/T-A7670X/issues/25#issuecomment-1346243254

deeja, 2023-07-26:
The GPS antenna that came with the device isn't working as well as other antennas I have. 
I have swapped the antenna and now the GPS fix happens in seconds.

ShonkyCH, 2023-07-27:
> I have swapped the antenna
Could you link the antenna you used for reference?

deeja, 2023-07-27:
It's from a local supplier, but as long as it has the `u.FL` connector, then you should be ok.
 Drone GPS antennas usually have that connection.

PequinhaSander, 2024-05-19:
Onde posso arrumar uma firmware do a7670sa open ?? Para fazer modificações

PequinhaSander, 2024-05-19:
Gostaria de modificar o nome da versão que aparece quando consulto pelo SMS com o comando version#, alguém pode me ajudar como consigo alterar ?