Xinyuan-LilyGO/LilyGo-Modem-Series #27: Problems with debugging code when the device is connected to computer

mp-mba, 2022-11-26:
Hi,

I've recently bought this device and I'm pretty happy with all the possibilities it gives me.
I have just one problem which is very annoying and it makes writing and debugging code very hard.

I'm using platformio and this example mainly: https://github.com/Xinyuan-LilyGO/T-A7670X/blob/main/examples/new_version/Arduino/MqttClient/MqttClient.ino

The problem is that device have constant problem connecting to mqtt broker when it's connected to computer and when I'm watching serial monitor.
When I detach the USB cable, reset the device it starting to work well and send metrics over mqtt (I have battery inserted).

Is my approach to program device wrong?

daveymg-nz, 2022-11-27:
Hi, I'm by no means an expert wit this but I've been doing similar coding. I suspect that when the USB is plugged in then it is the only power source, the battery is not used. The modem chip can have quite high current spikes so you may be experiencing brown outs if your pc usb port can't supply sufficient instantaneous power. I have run mine successfully from usb power only (no battery connected) and haven't experience any modem connectivity problems.

mp-mba, 2022-11-28:
Thanks. I've removed the battery, but it's behave the same.

daveymg-nz, 2022-11-28:
That would be correct as the battery is not used when the usb is connected. Maybe check the power available from your usb port.

mp-mba, 2022-11-29:
I've changed too many things at once to able to say what was the issue.
I've:
- rewrited the code,
- changed usb cable
- moved for windows to mac.
Looks like it's working now.

closed 2022-11-29 by mp-mba