Xinyuan-LilyGO/LilyGo-Modem-Series #29: Strange value from the GPS

peder2key, 2022-12-09:
Hello

I am testing the gps and getting back strange Lat and Lot values: 

at:-0.133333 lon:-0.000000
lat:-0.133333 lon:-0.000000
lat:-0.133333 lon:-0.000000
lat:-0.133333 lon:-0.000000
lat:-0.133333 lon:-0.000000
lat:-0.133333 lon:-0.000000
lat:-0.133333 lon:-0.000000
lat:-0.133333 lon:-0.000000
lat:-0.133333 lon:-0.000000
lat:-0.133333 lon:-0.000000
lat:-0.133333 lon:-0.000000
lat:-0.133333 lon:-0.000000
lat:-0.133333 lon:-0.000000
lat:-0.133333 lon:-0.000000
lat:-0.133333 lon:-0.000000
lat:-0.150000 lon:-0.000000
lat:-0.150000 lon:-0.000000
lat:-0.150000 lon:-0.000000
lat:-0.150000 lon:-0.000000
lat:-0.150000 lon:-0.000000
lat:-0.150000 lon:-0.000000
lat:-0.150000 lon:-0.000000
lat:-0.150000 lon:-0.000000
lat:-0.150000 lon:-0.000000
lat:-0.150000 lon:-0.000000
lat:-0.150000 lon:-0.000000
lat:-0.150000 lon:-0.000000

what is the unit? 

Best Regards

LilyGO, 2022-12-12:
Hello. Did you buy the one with GPS? Which example did you use?

peder2key, 2022-12-12:
Hello,

I bought this unit: https://www.aliexpress.com/item/1005003036514769.html?spm=a2g0o.order_list.order_list_main.156.1ca01802oLv0sw

I have tested this example: https://github.com/Xinyuan-LilyGO/T-A7670X/blob/main/examples/new_version/Arduino/TinyGSM_Net_GNSS/TinyGSM_Net_GNSS.ino

I have also tried to get the modem.getGPSraw() string also which gave me this result:

3,10,,00,00,5923.88109,N,01027.87128,E,091222,133005.00,67.9,2.319,,3.8,2.5,2.9

but the modem.getGPS(&lat, &lon) gives me: 
lat:-0.000000 lon:-0.000000

LilyGO, 2022-12-13:
It is caused by different parameter positions

miscawelo, 2022-12-15:
I bought the non GPS version and later connected an antenna I had laying around, but I can't power up the GPS, it erros out. Is there any difference between the two versions? Besides the included antenna

LilyGO, 2022-12-16:
> hmm what does that mean?

[reference](https://github.com/Xinyuan-LilyGO/T-A7670X/blob/main/examples/new_version/Arduino/TinyGSM_Net_GNSS/TinyGSM_Net_GNSS.ino#L284)

LilyGO, 2022-12-16:
> It's the difference without GPS. All other things

LilyGO, 2022-12-16:
> I bought the non GPS version and later connected an antenna I had laying around, but I can't power up the GPS, it erros out. Is there any difference between the two versions? Besides the included antenna

It's the difference without GPS. All other things

mikandrzej, 2023-11-18:
I also have issue with position of parameters. It seems that library has a bug - or maybe there should be command before using getGPS to change position of parameters...

lewisxhe, 2023-11-18:
Are you running the previous sketch or the latest clone?

closed 2024-01-02 by github-actions[bot]