Xinyuan-LilyGO/LilyGo-Modem-Series #30: Using Lora 868/915Mhz Shield with the A7670X

ivicajan, 2023-01-09:
I would like to know if this combination is possible? 
If it is should I order "a standard Lora Shield (https://www.lilygo.cc/products/ttgo-accessories-shield-lora-868-915-mhz)?

I have one spare  T-Higrow Lora Shield (http://www.lilygo.cn/prod_view.aspx?TypeId=50060&Id=1309&FId=t3:50060:3) and trying to make it work but no luck. I can't find RST and DI0 pins on the T-Higrow Lora.

Thanks in advance

LilyGO, 2023-01-10:
Hello，
 T-Higrow Lora Shield for T-Higrow .
 (https://www.lilygo.cc/products/ttgo-accessories-shield-lora-868-915-mhz)  for T-7000G. IO12 has been used in the A7670. Not fully compatible.

closed 2023-01-10 by LilyGO

ivicajan, 2023-03-21:
I have a separate standard LoRA module (RFM97) connecting with the standard pins to any ESP32:
RST  -> ?
CS    -> ?
SDO -> ?
SDI  -> ?
SCK -> ?
DIO1-> ?

I don't know which pins are free on the A7670X board so I can use them to connect the Lora module? 
If you could put the pin numbers instead of "-> ?" I wrote.

Thanks in advance!