Xinyuan-LilyGO/LilyGo-Modem-Series #32: HttpsClient example doesn't work

j0hnsmith, 2023-01-15:
When I run `/examples/new_version/Arduino/HttpsClient/HttpsClient.ino` it sends a http request to port 443, `The plain HTTP request was sent to HTTPS port`

```
[4667] Wait...
[7668] Initializing modem...
[7668] ### TinyGSM Version: 0.11.5
[7668] ### TinyGSM Compiled Module:  TinyGsmClientSIM7600
[12326] ### Modem: AT
[12326] ### Modem: AT
[22452] ### Unhandled: SMS DONE

+CGEV: EPS PDN ACT 1
[22452]  setNetworkMode faill
[22457] ### Modem: A7670E-LASE
[22457] Modem Name: A7670E-LASE
[22484] Modem Info: Manufacturer: INCORPORATED Model: A7670E-LASE Revision: A7670M7_V1.11.1 IMEI: 862205052859719 +GCAP: +CGSM,+FCLASS,+DS
Waiting for network... success
Network connected
Connecting to YourAPN success
GPRS connected
Performing HTTPS GET request... Response status code: 400
Response Headers:
    Server : nginx/1.10.3 (Ubuntu)
    Date : Sun, 15 Jan 2023 01:04:22 GMT
    Content-Type : text/html
    Content-Length : 280
    Connection : close
Content length is: 280
Response:
<html>
<head><title>400 The plain HTTP request was sent to HTTPS port</title></head>
<body bgcolor="white">
<center><h1>400 Bad Request</h1></center>
<center>The plain HTTP request was sent to HTTPS port</center>
<hr><center>nginx/1.10.3 (Ubuntu)</center>
</body>
</html>

Body length is: 280
Server disconnected
GPRS disconnected
```

Can you fix the example?

daveymg-nz, 2023-01-15:
I think you need to setup the SSL parameters in the modem to fully use https and this will be website dependent. The example code is mostly only changing the port number from 80 to 443.

h6w, 2023-01-28:
There doesn't seem to be an example of adding the root CA certificate.

I've tried using client->setCACert(rootCACertificate) but it says:
```
error: base operand of '->' has non-pointer type 'TinyGsmClient' {aka 'TinyGsmSim7600::GsmClientSim7600'}
     client->setCACert(rootCACertificate);
           ^~
exit status 1
base operand of '->' has non-pointer type 'TinyGsmClient' {aka 'TinyGsmSim7600::GsmClientSim7600'}

```

Middlerig, 2023-02-11:
I have a similar issue posting to an HTTPs server using the example in "A76XX Series_HTTP(S)_ApplicationNote_V1.03" It worked fine with a previous A7670 module but not the one on the T-SIM. I get error 715 "handshake". The only difference that I can see is that the T-SIM has version A7670M7_V1.11.1 and the working one on a waveshare module is A7670M7_V8.18.1.

How can I upgrade the firmware in the T-SIMs A7670 to be the same?

CyanoFresh, 2023-08-06:
Hi! I am choosing the board for MQTTS communication, will it work? I see you guys have a problem with https

Middlerig, 2023-08-06:
I was having trouble doing an HTTPS post but it turned out to be a faulty 
A7670E device. Another module worked perfectly. 
 
Richard

Virus-free.www.avg.com

CyanoFresh, 2023-08-06:
> I was having trouble doing an HTTPS post but it turned out to be a faulty 
> 
> A7670E device. Another module worked perfectly. 
> 
>  
> 
> Richard
> 
> 
> 
>  
> 
> 
> 
> ----- Original Message -----
> 
> From: Alex Solomaha ***@***.***>
> 
> Reply-To: Xinyuan-LilyGO/T-A7670X 
> 
> 
> To: Xinyuan-LilyGO/T-A7670X ***@***.***>
> 
> Cc: Middlerig ***@***.***>, Comment ***@***.***>
> 
> Sent: 06/08/2023 16:19:44
> 
> 
> 
> 
> 
> 
> 
> 
> Hi! I am choosing the board for MQTTS communication, will it work? I see you guys have a problem with https
> 
> 
> 
> 
> Virus-free.www.avg.com

oh, 40 bucks to find out that module is faulty 😭 have they refunded?
thank you for the reply!

Middlerig, 2023-08-06:
Ha! 20 was the first offer, refused and increased to 30. I would have expected a replacement T-sim. Even then I was days trying different code trying to see what I was doing wrong. Just not fun sometimes. Still I have a paperweight. RichardSent from my iPhone

aravibala, 2024-10-19:
Hi, I'm facing an issue using A7672S the same 715 handshaking error, i could not send a request to the server it is self-certified can anyone help me figure it out? how can I get a firmware for A7672S