Xinyuan-LilyGO/LilyGo-Modem-Series #33: Battery voltage problem

j0hnsmith, 2023-01-22:
Hi, I've just bought this board (in Jan 2023) because battery voltage can be read, the version I got (directly from LilyGo via aliexpress) is `2022-2-28 V1.1`. The pinout on the product page clearly shows that battery voltage can be read (see image below) yet #19 says that only `2022-7-28 V1.2` can read the voltage. 

[image]

The board I received doesn't have the functionality of the product I ordered, can you send me one that does have this functionality? How can I get in touch to discuss it?

j0hnsmith, 2023-03-24:
@LilyGO ?

theMladyPan, 2023-04-13:
I found out discrepancy between the voltage on pin35 and battery voltage. My advice is to use separate resistor divider and measure on another free pin.

lewisxhe, 2023-05-04:
Do you mean the one you received does not have ADC battery measurement function?
1. Please take photos of the front and back of the motherboard, and I will send it to relevant personnel for review
2. Use this [sketch](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/readBattery/readBattery.ino) to test whether the voltage can be read correctly

lewisxhe, 2023-05-24:
@j0hnsmith I just confirmed that V1.1 does not have adc voltage divider resistors
In this case you can only add the voltage divider resistor up and down 100K to GPIO35

j0hnsmith, 2023-05-31:
@lewisxhe I ordered the board from LilyGo in January 2023, the board was advertised at this time with ADC battery measurement, that's the functionality that I was buying. Why didn't I get sent a `2022-7-28 V1.2` board?

qnixAVA, 2023-08-06:
Hey, I have same problem with the V1.2, I connected it to ESPHOME at GPIO 32, and send me 1.033V, but my 18650 is at 3.5V

ShonkyCH, 2023-08-06:
>  Hey, I have same problem with the V1.2, I connected it to ESPHOME at GPIO 35, and send me 1.033V, but my 18650 is at 3.5V

Remember it will be scaled by the divider. It can't read up to 4.2V directly.

If only we had a schematic...

qnixAVA, 2023-08-06:
With the v1.2 we need divider too ?

ShonkyCH, 2023-08-06:
> With the v1.2 we need divider too ?

My understanding is earlier versions have no way to measure the battery voltage.

V1.2 adds a divider apparently but we don't know the values. There has to be some way to lower the voltage. You cannot connect a battery that is potentially charged to 4.2V to an ADC running on a 3.3V micro.

Based on the other thread reporting 1.033V for a 3.5V battery it's probably somewhere around 0.30. Something like 10k in series with 4.7k would be close. You should be able to measure them actually. I will try today on mine.

ShonkyCH, 2023-08-07:
Ok, by measuring in circuit, I see :

VV_BAT to IO35 = close enough to dead on 100k 
IO35 to GND = also close enough dead on 100k
VV_BAT to GND = close enough to 200k also.

NB: VV_BAT is *after* the switch so not measuring directly at the battery (V_BAT) and would not measure the battery being charged if the switch was off. Also measuring in circuit is not ideal but the fact that they are so both so close to 100k suggests they are accurate. It does take a few seconds for the multimeter to stabilise probably due to some capacitance but I'm convinced it is correct.

The values make sense being 1:1 divider and would be about what I'd pick for a 4.2V battery to not over voltage the ESP32 pins. So at 4.2V you should see about 2.1V at IO35.

They don't line up with the value posted of 1.033V for 3.5V but I'd need to validate that with some code of my own. I will do some experiments when I get a chance.

lewisxhe, 2023-08-07:
I updated the example for reading the battery voltage, check out [battery_voltage](https://github.com/Xinyuan-LilyGO/T-A7670X/blob/main/examples/battery_voltage/battery_voltage.ino), some instructions added above the sketch

ShonkyCH, 2023-08-07:
> @lewisxhe The board you advertised included battery voltage, the board I received doesn't have that. How do I return the board I have and receive the board I ordered?

I get where you're coming from, but may be just easier to add yourself? I don't think it's any more complex than 100k:100k resistive divider.

theMladyPan, 2023-08-07:
> @lewisxhe The board you advertised included battery voltage, the board I received doesn't have that. How do I return the board I have and receive the board I ordered?

I believe 100k:100k divider is far more simple and cheaper than returning and waiting for a new board. Please don't milk the manufacturer for such trivial mistake.

j0hnsmith, 2023-08-07:
> I believe 100k:100k divider is far more simple and cheaper than returning and waiting for a new board. Please don't milk the manufacturer for such trivial mistake.

I have no idea what a _100k:100k divider_ is or how I'd use it to find out the voltage. Yes I could learn/find out but this would not be simple for _me_ (although it sounds like it would be simple for you). 

I'm certainly not trying to _milk_ the manufacturer, just get what I ordered. It's been over 6 months and I still can't use the board for the project I bought it for. I don't feel like I'm being unreasonable, actually extremely patient.

theMladyPan, 2023-08-07:
I didn't meant to be rude. You need 2 resistors each of value 100kOhm. They cost like 5 cents and they have it in every DIY electronic store. 
[image]
now connect first resistor on Battery + terminal and pin35 (IO35), second resistor connect on battery - terminal and pin35 (IO35). You will need solder station to solder them. This is exactly what is missing on the board so it is 100%fix. This is fix which takes ~5min to do. Hope that helps. Stanley

qnixAVA, 2023-08-07:
> I updated the example for reading the battery voltage, check out [battery_voltage](https://github.com/Xinyuan-LilyGO/T-A7670X/blob/main/examples/battery_voltage/battery_voltage.ino), some instructions added above the sketch

@ShonkyCH did u test this? thats work ? 
Still answer me 0,38V instead of 3,5v

j0hnsmith, 2023-08-07:
> now connect first resistor on Battery + terminal and pin35 (IO35), second resistor connect on battery - terminal and pin35 (IO35). You will need solder station to solder them

@theMladyPan Thanks for providing detailed instructions, indeed that does sound simple, I'll try it. I assume to get the voltage I read from pin35 as per the [battery voltage](https://github.com/Xinyuan-LilyGO/T-A7670X/blob/main/examples/battery_voltage/battery_voltage.ino)? I really appreciate the help

ShonkyCH, 2023-08-07:
> @ShonkyCH did u test this? thats work ?
> Still answer me 0,38V instead of 3,5v

I did try this afternoon but I seem to have programming problems with mine. I can see the boot message and it goes into programming mode if IO0 is low. I haven't done much with it yet but have programmed it a few times previously. Even then it was problematic to program. So I have to figure that out.

theMladyPan, 2023-08-07:
> > now connect first resistor on Battery + terminal and pin35 (IO35), second resistor connect on battery - terminal and pin35 (IO35). You will need solder station to solder them
> 
> @theMladyPan Thanks for providing detailed instructions, indeed that does sound simple, I'll try it. I assume to get the voltage I read from pin35 as per the [battery voltage](https://github.com/Xinyuan-LilyGO/T-A7670X/blob/main/examples/battery_voltage/battery_voltage.ino) sketch? I really appreciate the help

yes, the voltage will be exactly (+/- precision of ADC/resistors) half of the Battery's voltage. If bat is 4.2V you should read 2100mV +/- (noise+precision). Again, this is how manufacturer do it on v1.2 of the board (as far as I know and remember)

ShonkyCH, 2023-08-07:
Note the divider is after the slide switch on v1.2. Main issue direct on the battery being that even when off you'll have a small load on the battery which will eventually kill if left too long. It is only ~20uA from the 200K. There may be a bit more through the CPU pin but can't be more than 40uA through the top half of the divider.

Actually calculating it, it is ****very**** small relative to the battery capacity and probably won't make a huge difference.

qnixAVA, 2023-08-07:
I just tried the new battery_voltage.ino sketch but they keep send me :1,033V but my multimeter tell me 3,49V. I don't know where is the problem. My battery is an 18650 3,7V. And I have the A7670E V1.2

ShonkyCH, 2023-08-08:
Measure the voltage at GPIO35 directly on the ESP32.

qnixAVA, 2023-08-08:
sensor:
  - platform: adc
    pin: 35
    name: "Battery Voltage"
    update_interval: 10s
    id: battery_voltage
    
    That's what I do, if I had:
 filters:
      - multiply: 3.37

I have +/- 3,5v, the correct value from my multimeter

ShonkyCH, 2023-08-08:
But you've just hacked in a scaling factor without understanding it. If it works, great but you can't be sure that's the problem.

Put a fully charged battery or a bench supply at 4.2V and see what it reads.

qnixAVA, 2023-08-08:
I have nothing to supply at 4.2V. 
I know that it's not the good solution but that's the only one I found to read a "correct" voltage

ShonkyCH, 2023-08-08:
Charge the battery? Are you using an 18650?

ShonkyCH, 2023-08-08:
So like I said, your multiplier is probably incorrect. There's something wrong with your device or program. If you have no method to even vary voltage how can you even properly test it?

I managed to get mine to reprogram (seems Q9 to pass the ISP_TX to UAR0_RX is faulty on mine - I just bridged it with tweezers and it worked straight away - annoying but should be fairly simple to fix).

Using a Nordic Power Profile connected to the battery terminals:

|Nordic PPK Output (V) | Measured At GPIO35 (V) | Reported By Sketch (V) |
| --- | --- | --- |
| 3.3V | 1.615V | 3.32V |
| 3.5V | 1.711V | 3.53V |
| 3.7V | 1.813V | 3.76V |
| 3.9V | 1.911V | 3.98V |
| 4.1V | 2.013V | 4.20V |
| 4.2V | 2.062V | 4.30V |
| 4.3V | 2.106V | 4.40V |

I wouldn't trust the PPK output so much as it's through some smallish connections so there may be a bit of voltage drop I haven't accounted for.

Other than that whilst it's not very linear it basically works. It mostly reads about 100mV high in my case which isn't great but I expect I can make that a bit better (or simply apply an offset to calibrate it out).

I also got this output detected by the sketch. I haven't looked into what this means exactly yet though. I see it assumes varied reference voltages when calculating the actual value to output.
`eFuse Vref:1149 mV`

qnixAVA, 2023-08-15:
> Measure the voltage at GPIO35 directly on the ESP32.

The voltage on GPIO35 on ESP is 1,7v
My 18650 is at 3,45V .. What is wrong?
I got V1.2

theMladyPan, 2023-08-15:
> > Measure the voltage at GPIO35 directly on the ESP32.
> 
> The voltage on GPIO35 on ESP is 1,7v My 18650 is at 3,45V .. What is wrong? I got V1.2

As mentioned above, there is a 100k/100k resistor divider because ESP32 can read only up to 3.3V. You always need to multiply your reading by 2.

qnixAVA, 2023-08-15:
> 

Yep, so why when I code this :  
sensor:
 - platform: adc
    pin: 35
    name: "Battery Voltage"
    update_interval: 10s
    id: battery_voltage
    filters:
      - multiply: 2

in esphome.

That send me: 'Battery Voltage': Sending state 2.06600 V with 2 decimals of accuracy

ShonkyCH, 2023-08-16:
Because whatever you are running on the Esp32 is doing the A to D conversion wrong or applying a wrong scaling factor.

closed 2023-12-08 by github-actions[bot]