Xinyuan-LilyGO/LilyGo-Modem-Series #41: GPS/GNSS does not work

steven-redkitedesign, 2023-04-14:
Using Lilygo T-A7670E board, GNSS does not work.
Tried the test method described here: https://github.com/Xinyuan-LilyGO/T-A7670X/issues/25

Result:
08:42:42.014 -> $GNGGA,064243.00,,,,,0,00,99.99,,,,,,*7F
08:42:42.048 -> $GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
08:42:42.117 -> $GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
08:42:42.151 -> $GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
08:42:42.220 -> $GPGSV,1,1,03,19,,,20,24,,,30,17,,,21,0*6C
08:42:42.253 -> $GAGSV,1,1,00,$GNRMC,064244.00,V,,,,,,,140423,,,N,V*19

So the time is correct (but perhaps that is from the GSM side, as 4G is working) but there is no location.

Tried outside, waited for >1hr so it could get almanac, makes no difference.

Anonymouse-X, 2023-04-26:
same here this is the result for me

OK
AT+CGNSSPWR=1

OK

+CGNSSPWR: READY!
AT+CGNSSMODE=3

OK
AT+CGNSSNMEA=1,1,1,1,1,1,0,0

OK
AT+CGPSNMEARATE=2

OK
AT+CGNSSTST=1

OK
AT+CGNSSPORTSWITCH=0,1

OK
$GNRMC,061609.50,V,,,,,,,260423,,,N,V*15
$GNGGA,061609.50,,,,,0,00,99.99,,,,,,*75
$GNGLL,,,,,061609.50,V,N*59
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,2,1,05,06,,,27,14,,,27,19,,,27,194,,,29,0*5A
$GPGSV,2,2,05,195,,,28,0*57
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
$GNRMC,061610.00,V,,,,,,,260423,,,N,V*18
$GNGGA,061610.00,,,,,0,00,99.99,,,,,,*78
$GNGLL,,,,,061610.00,V,N*54
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,2,1,05,06,,,30,14,,,26,19,,,29,194,,,32,0*59
$GPGSV,2,2,05,195,,,28,0*57
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E

lewisxhe, 2023-04-27:
1. Please confirm that the following two commands have been run before running GPS positioning, it will turn on the power supply of the GPS antenna, otherwise GPS positioning will be invalid

```
AT+CGDRT=4,1
AT+CGSETV=4,0
```

2. After executing the above command, check whether there is 2.5~3V power supply at the antenna end.
[image]
3. Positioning must be done outdoors, and positioning cannot be performed indoors.

Anonymouse-X, 2023-04-29:
Yes! there is indeed voltage. I read about 3.06V but the output in the serial monitor is the same as before.

lewisxhe, 2023-04-30:
If all the conditions are met and it is still impossible to locate, you can consider whether there is an antenna issue. Do you have any other antennas to try? Being able to output valid NMEA sentences indicates that GNSS is working normally

lewisxhe, 2023-07-27:
If you still have problems, please reopen the issue

closed 2023-07-27 by lewisxhe