Xinyuan-LilyGO/LilyGo-Modem-Series #56: How to update "GPSShiled_T-A7670X_20230613.bin" to T-A7670G ?

booroono, 2023-08-10:
How to update "GPSShiled_T-A7670X_20230613.bin" to T-A7670G ?

lewisxhe, 2023-08-10:
This firmware is for an external GPS module. Are you sure you have a GPS on your board?

booroono, 2023-08-10:
I have the T-A7670G model. 
So, does the T-A7670E support GPS?
Please check.

lewisxhe, 2023-08-10:
no support

lewisxhe, 2023-08-10:
Check your order and send it to customer service for confirmation

booroono, 2023-08-10:
I already left about this on "www.lilygo.cc"
What does external GPS module mean?
pls show me the image.
However, the 7670G and 7670E have a GPS antenna socket.

lewisxhe, 2023-08-10:
[image]

booroono, 2023-08-10:
Thanks lewisxhe,
but 
[image]
I'll try to find another module.
thanks again.

closed 2023-08-21 by booroono