Xinyuan-LilyGO/LilyGo-Modem-Series #57: T-A7670G GPS not working.

booroono, 2023-08-10:
Hi all, 
I have two T-A7670Gs.
The HW version is V1.2 (2022-7-18).
[image]
I want to use real time GPS tracking.
But the GPS doesn't work.
[image]
additional information as below.
---------------------------
AT+SIMCOMATI

Manufacturer: INCORPORATED
Model: A7670G-LLSE
Revision: A011B15A7670M7
A7670M7_B15V01_230111
---------------------------

lewisxhe, 2023-08-10:
7670G  nosupport gps

ShonkyCH, 2023-08-10:
> 7670G no support gps

You should fix your AliExpress listing then:

[image]

closed 2023-08-10 by booroono

lewisxhe, 2023-08-10:
Check your order and send it to customer service for confirmation