Xinyuan-LilyGO/LilyGo-Modem-Series #58: SD Card triggers Brown-out, never works

i-e-b, 2023-08-10:
Hi.

I've spent about 2 days trying to get my Lilygo T-SIM A7670E to read an SD card.

- I have tried a total of 8 different SD cards, ranging from 256MB to 32GB, from a range of different manufacturers.
- I have tried different power supplies: USB, Bench-top power supply at 3.3v, Power supply into solar input at 5V; All of the above with and without an 18650 cell in the battery holder.
- I have progressively turned off as many features as possible on the ESP32, and powered off the SIMCOM module
- I have tried every "how to" and sample I can find, including all the Espressive examples
- I have tried the Arduino/Platform.IO `SPI.begin(...)` and `SD.begin()` method
- I have tried every ESP IDF interface I can find, from the high level vfat, to low level SPI and SDMMC interfaces

About 1 in 100 times, an SD card will be recognised, and will have capacity read, but the SD card can't actually be used.
The rest of the time, the warning `Brownout detector was triggered` will show, and the device will reboot.

I have tried disabling the brown-out detector -- the device will no longer reboot, but 100% of attempts to read the SD card fail.

I have tried 4 different Lilygo T-SIM A7670E boards, and they all show the same issue.

Can you please tell me, what is the EXACT process to get an SD card working, including the EXACT setup you use to test?

ShonkyCH, 2023-08-11:
Doesn't really help but have you monitored VDD3V3? Just put a scope on 3V3 header pin (PY pin 16). Then presuming you see a large voltage drop I'd then go looking at the input to U14 although that will be harder to find on the board with no layout and minimal visible silkscreen.

If it's really browning out you really need a solution for that first IMO. If it's that close to the limit any other things you might do are going to be band aids at best.

i-e-b, 2023-08-11:
No significant drop that I can see. 
Brownout detector is set for 2.43V +-0.05. I seem to be well above this even in the worst case.

Testing a read cycle of SD card. All of these with Modem, Wifi, BT off:

- **Battery only:** stable at `2.8V`
- **Battery and 5V into solar:** stable at `3.1V`
- **5V into solar only:** bouncing between `2.8V` and `3.0V`
- **Directly powering:** stable `3.3V` (although I'm really reading the power supply with this setup)

Ripple seems to be around `0.2V` at the microsecond scale.
The 18650 has about 3 months of use, and is kept fully charged (open `4.0V`)

For reference, with battery and 5V into solar:
- Programming ESP32 (flash clear & write) -> `2.8V`
- UDP traffic on modem -> `2.8V`

ShonkyCH, 2023-08-11:
You're saying your 3.3V rail is 2.8V? That's not right.

i-e-b, 2023-08-11:
Thanks for your time on this.

I'm mainly running this with 5V to the solar, and a battery in, so around 3.1V -- which does still seem low.
I have two boards with me -- but have tested with another 2 (they've gone to a colleague now)

Minimum sketch that fails for me:

```
#include <Arduino.h>
#include <SPI.h>
#include <SD.h>

void setup() {
  Serial.begin(115200);
  SPI.begin(/*SD_SCLK*/14, /*SD_MISO*/ 2, /*SD_MOSI*/ 15, /*SD_CS*/ 13);
  if (!SD.begin(13))
  {
    Serial.println("SDCard MOUNT FAIL");
  }
  else
  {
    uint32_t cardSize = SD.cardSize() / (1024 * 1024);
    String str = "SDCard Size: " + String(cardSize) + "MB";
    Serial.println(str);
  }
}

void loop() {
  delay(1000);
}
```

My main project is in ESP-IDF.

i-e-b, 2023-08-11:
Powering from USB gives me a super-stable `3.2V` with barely any ripple -- but that's not going to be particularly helpful in the field.

ShonkyCH, 2023-08-11:
I do not see the ripple you see, nor the low voltages. VDD3V3 is the direct output from the 3.3V linear regulator. It should really be bang on 3.3V and mine is. I see no glitches or significant ripple. The noise is of the order of less than +/- 10mV that I see whilst running off an 18650 and right on 3.3V

I tried a 16GB Kingston and a 32GB "Tandy" (not really Tandy so consider it no name brand). They both always work when powered by USB-C viewing the terminal in the Arduino Serial Monitor.

Your sketch however will not work running off battery/solar. The V3V rail to the SD card is switched on by either VBUS (from the USB-C power) or from IO12. So running on battery/solar requires IO12 to be high. Once I added that, the sketch worked fine running off a fully charged 18650 (about 4.15V). This is using a logic analyzer to catch the output from the TXD line as you can't have the USB-C connected or it will "override" the battery supply (and also force on the FET switching V3V). 

I was going to look for V3V specifically before I noticed it was switched, however it it will be slightly hard to find without proper PCB layout information.

Let me know if you want to try anything else.

i-e-b, 2023-08-11:
Interesting. My scope is terrible, so it could be that it's finally on its death-bed. I would think if the power on my boards was that much worse I'd be seeing other issues too.

I've got `IO12` noted as the power to the modem.
Circuit diagram doesn't show `IO12` connected to SD card slot.

I will have a play, assuming that I've got something wrong in here.
Thanks again.

[image]

ShonkyCH, 2023-08-11:
The schematic is quite poorly drawn, upside, backwards from convention. V3V is not the same as VDD3V3 which powers the ESP32. It is switched as per my last post:

[image]

i-e-b, 2023-08-11:
Ah, this is great! Many thanks -- things are a lot clearer now.

So I've been powering off the entire board when not using the modem! 😅

I can now detect the SD card reliably. 

Thank you again!

ShonkyCH, 2023-08-11:
> So I've been powering off the entire board when not using the modem! 😅
> 
> I can now detect the SD card reliably. Somewhat amazed that it sometimes worked before!

Not the entire board. Just the modem and the SD card slot mainly. It would also disable the link between the USB-C UART and the ESP32 if V3V is off. Not sure that disabling the comms that way is all that great.

The reason it kinda worked is probably the SD card was powering itself through protection diodes on its IO from the ESP32 GPIO lines. Yes that would not be reliable.