Xinyuan-LilyGO/LilyGo-Modem-Series #60: shut down modem

rgusso, 2023-09-12:
hi, i'm using the example and i'm able to start the modem using this part:

    //A7670 Reset
    pinMode(RESET, OUTPUT);
    digitalWrite(RESET, LOW);
    delay(100);
    digitalWrite(RESET, HIGH);
    delay(3000);
    digitalWrite(RESET, LOW);

    pinMode(PWR_PIN, OUTPUT);
    digitalWrite(PWR_PIN, LOW);
    delay(100);
    digitalWrite(PWR_PIN, HIGH);
    delay(1000);
    digitalWrite(PWR_PIN, LOW);

But now i want to turn it off , how do i do?

HDC67, 2023-09-13:
Have you looked at the AT command reference?

```AT+CPOF```

Or the Chinese datasheet shows a 2.5 sec pulse for Toff in section 3.2.2

metarsul, 2023-09-15:
` Or the Chinese datasheet shows a 2.5 sec pulse for Toff in section 3.2.2 `

This did the trick
Thank you

closed 2023-09-15 by rgusso