Xinyuan-LilyGO/LilyGo-Modem-Series #61: Enabling BAT(IO12) causing brown out

scsPEp, 2023-09-13:
Hello,
I am currently using the T-A7670E V1.2 and I am facing the problem that the supply voltage takes a big dip when I enable the signal BAT(IO12), which causes the ESP to brown out. After the brown out reset enabling the signal will work, as the capacitors are already charged then. At least that is my guess. Is this a known problem?

The problems stays the same when using a 5A power supply instead of a battery.

This is annoying to me, as my device is most of the time in deep sleep to save energy, and should enable the modem only on some wake ups. I am using the RTC memory to keep some information between wake ups, but everytime I want to turn on the modem a brown out happens at least once and I lose the RTC memory.

HDC67, 2023-09-13:
I don't seem to have this issue when powered by just the USB using the example modem application.

Do you have a minimum code example the demonstrates the problem?

scsPEp, 2023-09-14:
If you power the device via USB the BAT signal will always be active, this can be seen on page 3 of the schematic, in the bottom right corner. VBUS is used parallel to IO12. So I do think there should also be a dip on the voltage, but only once when you plug in the device. As long as the BAT signal is switched on this problem should not happen again. But using a battery and keeping IO12 on during deep sleep will drastically increase battery consumption from what I have seen.
The following measuring was taken while switch on the BAT signal. The blue signal is IO12 used to switch on BAT, the yellow signal is VDD3V3. Red signal is IO4 to switch PWRKEY and the green signal is IO5 to control modem RESET. As you can see, as soon as IO12 goes to high there is a dip, This measurement did work without a brown out, but usually the voltage will dip enough to cause a brown out.

[image]

HDC67, 2023-09-14:
Right you are. That's pretty bad then though if it's enough to brown out the ESP32. It basically means it's difficult to power up the modem on battery.

I haven't really had much chance to play with mine but I think I also did have it running on an 18650 and then going to deep sleep including turning off IO12 then waking up and turning the modem on. I didn't measure the ESP32 supply when doing that so it may have been dipping like yours but not enough to cause a brown out.

scsPEp, 2023-09-14:
I do think it can also be easily missable if you are not using something like the RTC memory or connect a terminal to the UART and enable debug output, as it will start up after 1 or more brown out cycles. I started to wonder when I was receiving 0 on values stored in RTC memory which should not be 0 after the deep sleep, as they should be set to something else from the previous wake cycle. Right now I changed my implementation to use NVS instead of RTC memory to keep those values, but this problem still is annoying, as needing multiple boots after wake up also consumes more battery.

neuralprocessing, 2023-09-25:
Has there been any solution or hint, how to prevent this reboot/brown out in between?
I also would like to achieve the minimal power consumption as possible to run my device as long as possible without recharging it.

neuralprocessing, 2023-09-25:
Hi - I am using the same T-A7670E V1.2. The RTC Memory is not being deleted, so the deep sleep and wake do not seem to burn out the esp32. I was using a slightly modified version of this:
https://randomnerdtutorials.com/esp32-external-wake-up-deep-sleep/

scsPEp, 2023-09-25:
Hi, are you using the modem in your examples? On my device using the deep sleep and wake up on it's own is also working fine. When I enable the modem supply via IO12 my device goes into brown out. Most likely because of the capacities connected to the modem on the supply pin. Is this also working for you without a problem?

neuralprocessing, 2023-09-25:
Yes - I am using the modem. I send myself back the boot number and the reason for wake up. BAT_EN 12 pulled to HIGH on startup.

closed 2023-12-06 by github-actions[bot]

Rikests, 2024-02-14:
I have 14 T-A7670E boards. There are a few that do not brown out when I pull IO12 high, some brown out often, and some that brown out every time. I tried disabling brownout detection, but that often caused the ESP32 to hang and become unresponsive until battery dies. 
Now I turn on the IO12 at the start of the setup{} and soldered a cap on ESP32 power pin. This seems to work better, but some of the boards still brown out sometimes.
I will do more testing to see how to improve this situation, but I really hoped LilyGO would not have such basic issues as power delivery.

jimbobshocks, 2024-09-29:
First off, the T-A760E is a great little board. Well done Lilygo, and do keep improving it!

I have this same issue with my two T-A7670E boards I am evaluating. It is a hard problem to understand as it does not happen when the board is powered by USB during development, but only when it is powered by the 18650 battery. If you are not using RTC memory, or requiring a bootcount, then you probably won't see it.

The symptoms of the brownout after setting IO12 - BOARD_POWERON_PIN HIGH to power the board and modem:

1. RTC memory is reset losing all stored data
2. RTC is not affected... the clock is not interrupted
3. The brownout and restart only happens once, the first time the board is started after deepsleep. The second boot and board powerup is fine.

My boards were bought directly from lilygo, and are 2022-7-18 v1.2 A760E version

An example with modem info for tracing the board via the modem info if necessary:

Modem Name: A7670E-LASE
Modem Info: Manufacturer: INCORPORATED Model: A7670E-LASE Revision: A7670M7_V1.11.1 IMEI: 867255075558454 +GCAP: +CGSM,+FCLASS,+DS

I have not tried @Rikests solution, though it looks workable. I am working around this issue by updating the RTC from a remote config service after the modem connects, and hence kind of making it persistent, and on most of my boots, I don't use the modem.

Other fun software hacks, are after wakeup from deepsleep when you are going to use the modem, store the RTC values in SPIFFS, power the board, brownout and survive, write the RTC values back to memory. Bit painful, but avoids soldering.

jimbobshocks, 2024-09-29:
@lewisxhe Hi! I can see you are actively providing support to this product, and doing a very good and patient job ;) It looks like this is a real issue. Is there any plans for a v1.3 of the hardware to resolve it?

lewisxhe, 2024-10-01:
I think it's not a good idea to have io12 control the modem power. There are no plans to do so at the moment.

jimbobshocks, 2024-10-01:
@lewisxhe What is the reason why setting IO12 - BOARD_POWERON_PIN high causes this brownout issue? I have not traced the circuit for the BOARD_POWERON_PIN, but it seems to control power to the modem in this board. If you don't set BOARD_POWERON_PIN high, the modem does not work. If I am not understanding this, please confirm what BOARD_POWERON_PIN actually does.

lewisxhe, 2024-10-04:
[image]

I drew a simple diagram. When IO12 is high, Q8 is in the on state. If IO12 is low, Q8 is in the off state.

jimbobshocks, 2024-10-10:
@lewisxhe Thank you for sharing the drawing and explaining it. It is a clever design as using IO12 to turn on the board board, to control the modem and other devices like level shifters etc means that the deep sleep power consumption of the A760E is very low (it will run for a year on a single battery if you don't use the modem too much and sleep most of the time). Do you agree with what @scsPEp  posted on Sept 14, 2023?

This makes sense to me, as what I see is that the ESP32-WROVER soft resets when the Lilygo-A760E is powered by battery, at the point where IO12 is set high to power the board including the modem on Q8. This seems like a brownout. I wish this did not happen as it means you can't use the RTC memory when sleeping.