Xinyuan-LilyGO/LilyGo-Modem-Series #62: Pins avaible

metarsul, 2023-09-13:
Hello I am new to this world of esp32.
I developed a program using a simple development board where I configured Serial1 to connect to an RS485 module using pins 16 and 17
I saw that many pins on this module are reserved and in many examples they use serial1 for AT commands. What pins can I use and can I declare serial1 in the code to use them.

HDC67, 2023-09-13:
[image]

And check the schematic to see which pins have net labels that are used. Some of them like IO12 are used to enable power to the modem as well as being available on the module headers

metarsul, 2023-09-25:
i was able to use pins 32 and 34

closed 2023-09-25 by metarsul