Xinyuan-LilyGO/LilyGo-Modem-Series #63: Example for MQTTS connection

caeb92, 2023-09-23:
Following the examples  from the repository I have been able to establish an MQTT connection with rabbitMQ using port 1883 with user and password settings, i can publish and consume messages from topics without problem, but when trying to connect to hivemq cloud through the port 8883 with SSL the connection fails, 

Im using TinyGsmClient and PubSubClient for establish connections with MQTT brokers, i was reading the documentarion of [TinyGsmClient](https://github.com/vshymanskyy/TinyGSM#supported-boardsmodules) and says SSL/TLS(HTTPS) Are not yet supported on: Quectel modems, SIM 5360/5320/7100, SIM 7500/7600/7800.

I was wondering if there is an easy way to establish the connection or I must use AT commands?

CyanoFresh, 2023-09-25:
also would like to get an example

lewisxhe, 2023-11-28:
I have added an example of MQTTS

closed 2024-01-13 by github-actions[bot]