Xinyuan-LilyGO/LilyGo-Modem-Series #65: Power source detector

Unicorn7141, 2023-09-29:
Hello there,
I want to make a device that alerts the user when the device's power is off (ka running from battery).
How can I do that, because I assume `VBAT` is not directly readable and I am not sure if it outputs 3.3v or 5v when running from USB.

HDC67, 2023-10-05:
Connect your supply (i.e. ```SOLAR_IN``` with an appropriate voltage divider to a CPU pin?

VBAT is readable on pin 35 if you have version 1.2 (or later of the board)
https://github.com/Xinyuan-LilyGO/T-A7670X/issues/33

Unicorn7141, 2023-10-13:
> Connect your supply (i.e. `SOLAR_IN` with an appropriate voltage divider to a CPU pin?
> 
> VBAT is readable on pin 35 if you have version 1.2 (or later of the board) See this issue : #33

I have V1.1

HDC67, 2023-10-15:
> > Connect your supply (i.e. `SOLAR_IN` with an appropriate voltage divider to a CPU pin?
> > VBAT is readable on pin 35 if you have version 1.2 (or later of the board) See this issue : #33
> 
> I have V1.1

You could add your own with a simple 100k:100k voltage divider fed back into a pin capable of doing AD conversion. That's all that is on the v1.2 as far as we know (there's no v1.2 schematic but that is what the manufacturer says and what measurements appear to show)