Xinyuan-LilyGO/LilyGo-Modem-Series #68: 4G Support

Flamabalistic, 2023-10-12:
Does this library/ board support 4G communication? 

I believe the A7670 itself supports it, however I can't see any examples which use it. Any info would be much appreciated!

HDC67, 2023-10-12:
As long as the network/SIM/bands match your carrier then yes. What do you think needs to be done specially? There's very little to ever be done. You can usually disable 4G or 3G if anything but that's about all is ever necessary and only if you need to disable it.

I have found the APN does need to be set for LTE to work on a commercial project.

Flamabalistic, 2023-10-12:
I haven't done too much in the cellular space and wanted to be sure before buying, thanks for the quick response!

closed 2023-10-12 by Flamabalistic

HDC67, 2023-10-12:
Note: I'm just a user like you.

Just make sure you buy the right modem version for your country that supports the appropriate bands.