Xinyuan-LilyGO/LilyGo-Modem-Series #69: question about ADC 5V

booroono, 2023-10-26:
Can I use 5V ADC to external from A-T7670G?

HDC67, 2023-10-27:
Explain further. You could use a 5V based ADC if you have appropriate level shifting on the SPI/I2C/whatever interface lines.

boseonlee, 2023-10-27:
What happens if I connect 5V ADC input directly to the ADC pin of A-T7670G?

HDC67, 2023-10-27:
> What happens if I connect 5V ADC input directly to the ADC pin of A-T7670G?

You don't connect an ADC input to an ADC pin. What you're asking doesn't make sense.

If you're asking if you can put a 5V signal on an ADC input (or any input for the matter) of the ESP32 then no you should not do that.

Use a voltage divider e.g. 100k//100k to divide the 5V signal down to 2.5V.

You should not apply voltages greater than the IO voltage of any device unless it specifically says it is tolerant of higher voltages.

closed 2023-12-11 by github-actions[bot]