Xinyuan-LilyGO/LilyGo-Modem-Series #71: question about T-SIM A7670E

asat232, 2023-11-17:
I need to make voice calls using microphone as input, is it possible? 
On the schematic are shown MIC+ and MIC-, but i dont see  them on pinout diagram. 
Can i connect microphone to some pads on the board, or on board is present built-in microphone?

lewisxhe, 2023-11-18:
Yes, you need to connect the condenser microphone wires to MIC+, MIC-

asat232, 2023-11-18:
looking the picture, as it seems to me MIC+/MIC- and SPK+/SPK- contacts on the side of the battery holder, close to its bottom on the right. Am i right?
(i dont have the board at the moment, just considering to order it and use it for my new project)

lewisxhe, 2023-11-18:
> looking the picture, as it seems to me MIC+/MIC- and SPK+/SPK- contacts on the side of the battery holder, close to its bottom on the right. Am i right?
> (i dont have the board at the moment, just considering to order it and use it for my new project)

yes

asat232, 2023-11-19:
thanks a lot

closed 2023-11-19 by asat232