Xinyuan-LilyGO/LilyGo-Modem-Series #72: T-A7670G - Fail at call

SeeNguyen, 2023-11-20:
Hi,
Could you do a help?
I'm using your T-A7670G with (1) WebServer.h, ArduinoOTA.h, (2) 5.3V input, no battery; everything was okay before, now when I ask for call by web button, the A7670G's falling down as picture (2 leds're off)
[image]
I've tried your Voice calls example, it's the same, not call.

Please advise!

lewisxhe, 2023-11-20:
Can you attach your code? Keep only the call part

SeeNguyen, 2023-11-20:
My code
1. Declaration
``/*=== A7670G declare ===*/
#define TINY_GSM_MODEM_SIM7600  //A7670 Compatible with SIM7600 AT instructions
#define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb
#define SerialAT Serial1
// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon
// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial
// See all AT commands, if wanted
//#define DUMP_AT_COMMANDS
#include <TinyGsmClient.h>
//#include <SPI.h>
//#include <SD.h>
#include <Ticker.h>
TinyGsm modem(SerialAT);
#define UART_BAUD    115200
#define PIN_DTR      25
#define PIN_TX       26
#define PIN_RX       27
#define PWR_PIN      4
#define BAT_ADC      35
#define BAT_EN       12
#define PIN_RI       33
#define RESET        5
//#define SD_MISO     2
//#define SD_MOSI     15
//#define SD_SCLK     14
//#define SD_CS       13

char mgsm[25];
char number[] = "+82-";//Change the number you want to dial
/*=== END A7670G declare ===*/

/*=== OWN coding ===*/
#include <HTTPClient.h>
#include <WiFiClient.h>
#include <Arduino.h>
#include <WiFi.h>
#include <WebServer.h>
WebServer webServer(80);
#include <ArduinoOTA.h>
#include <WiFiUdp.h>
WiFiUDP ntpUDP;
//NTP time
#include <NTPClient.h>``

SeeNguyen, 2023-11-20:
> dfcdf

SeeNguyen, 2023-11-20:
My code
1. Declaration
> /*=== A7670G declare ===*/
> #define TINY_GSM_MODEM_SIM7600  //A7670 Compatible with SIM7600 AT instructions
> #define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb
> #define SerialAT Serial1
> // Define [...]

3. In VOID SETUP
> void setup()
> {
>   pinMode(light, OUTPUT); digitalWrite(light, relayOFF); //HIGH is cut power to off the module led
>   pinMode(horn, OUTPUT); digitalWrite(horn, relayOFF); //abv
>   pinMode(buzzPin, OUTPUT); //LOW
>   pinMode(rfPin, INPUT); //LOW
> 
>   Serial.begin(UART_BAUD);
>   while (!Serial); delay(200);
>   connectWifi(); delay(1000);
>   connectOTA(); delay(1000);
> 
>   /*=== START A7670G Code ===*/
>   // Set console baud rate
>   SerialMon.begin(UART_BAUD);
>   delay(10);
>   SerialMon.println("setup...");
>   pinMode(BAT_EN, OUTPUT);
>   digitalWrite(BAT_EN, HIGH);
>   //A7670 Reset
>   pinMode(RESET, OUTPUT);
>   digitalWrite(RESET, LOW); delay(100);
>   digitalWrite(RESET, HIGH); delay(3000);
>   digitalWrite(RESET, LOW);
>   pinMode(PWR_PIN, OUTPUT);
>   digitalWrite(PWR_PIN, LOW); delay(100);
>   digitalWrite(PWR_PIN, HIGH); delay(1000);
>   digitalWrite(PWR_PIN, LOW);
>   delay(1000);
>   SerialAT.begin(UART_BAUD, SERIAL_8N1, PIN_RX, PIN_TX);
>   // Restart takes quite some time
>   // To skip it, call init() instead of restart()
>   SerialMon.println("Initializing modem...");
>   if (!modem.init()) {
>     SerialMon.println("Failed to restart modem, attempting to continue without restarting");
>   }
>   /*=== END A7670G Code ===*/
> 
>   /*=== START OWN Code ===*/
>   ArduinoOTA.begin();
> 
>   if (WiFi.waitForConnectResult() == WL_CONNECTED) {
>     timeClient.begin();
>   }
>   //  //Start Async Server for CLIENT Request
>   //  asyncServer.on("/ALARM", HTTP_GET, [](AsyncWebServerRequest * request) {
>   //    request->send_P(200, "text/plain", AlarmON().c_str());
>   //  });
>   //  //  asyncServer.on("/CareOFF", HTTP_GET, [](AsyncWebServerRequest * request) {
>   //  //    request->send_P(200, "text/plain", CareOFF().c_str());
>   //  //  });
>   //  asyncServer.on("/sysTEST", HTTP_GET, [](AsyncWebServerRequest * request) {
>   //    request->send_P(200, "text/plain", sysTEST().c_str());
>   //  });
>   //  asyncServer.begin();
> 
>   //Start webserver
>   webServer.on("/", handle_OnConnect); webServer.onNotFound(handle_NotFound);
>   webServer.on("/alarmTEST", handle_AlarmTEST); webServer.on("/callTEST", handle_callTEST);
>   webServer.on("/restart", handle_Restart);
>   webServer.on("/alarmON", handle_AlarmON); webServer.on("/alarmOFF", handle_AlarmOFF);
>   webServer.on("/config", handle_Config); webServer.on("/stat", sysSTAT);
>   webServer.begin();
> }

5. Function
> // Call Func
> void careCall() {
>   sprintf(mgsm, "D%S;", number ); delay(200);
>   modem.sendAT(GF(mgsm));
>   // check for any of the three for simplicity
>   //  if (modem.waitResponse(10000L) != 1) {
>   //    DBG("call faill");
>   //  }
>   Serial.println("call");
> }

lewisxhe, 2023-11-20:
There seems to be no problem. Can you use ATDebug to conduct dial-up testing alone?

SeeNguyen, 2023-11-20:
I'm trying with your VoiceCall example and remove my own libraries one-by-one to find the problem. I also try to downgrade TinyGSM to 0.10.x but not work.
In other hand, could you provide an example with WebServer?

SeeNguyen, 2023-11-21:
Just update after one night, this can make call with your clean VoiceCalls example, and ONLY make call wifi library. It's fail with the rest of libraries (just include only, not their functions).

Here're code: 
```
/**
   @file      Voice_Call.ino
   @author    Lewis He (lewishe@outlook.com)
   @license   MIT
   @copyright Copyright (c) 2023  Shenzhen Xin Yuan Electronic Technology Co., Ltd
   @date      2023-10-26
   @note      Not support T-SIM7672
*/

//#include "utilities.h"
#define TINY_GSM_MODEM_SIM7600
#define MODEM_BAUDRATE                      (115200)
#define MODEM_DTR_PIN                       (25)
#define MODEM_TX_PIN                        (26)
#define MODEM_RX_PIN                        (27)
// The modem boot pin needs to follow the startup sequence.
#define BOARD_PWRKEY_PIN                    (4)
#define BOARD_ADC_PIN                       (35)
// The modem power switch must be set to HIGH for the modem to supply power.
#define BOARD_POWERON_PIN                   (12)
#define MODEM_RING_PIN                      (33)
#define MODEM_RESET_PIN                     (5)
#define BOARD_MISO_PIN                      (2)
#define BOARD_MOSI_PIN                      (15)
#define BOARD_SCK_PIN                       (14)
#define BOARD_SD_CS_PIN                     (13)
#define BOARD_BAT_ADC_PIN                   (35)
#define MODEM_RESET_LEVEL                   HIGH
#define SerialAT                            Serial1

#define MODEM_GPS_ENABLE_GPIO               (-1)
// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// See all AT commands, if wanted
#define DUMP_AT_COMMANDS

char number[] = "+abc";//Change the number you want to dial
#include <TinyGsmClient.h>
TinyGsm modem(SerialAT);

#include <WiFi.h>
#include <HTTPClient.h>
#include <WiFiClient.h>
#include <Arduino.h>
#include <WebServer.h>
WebServer webServer(8080);
#include <ArduinoOTA.h>
#include <WiFiUdp.h>
WiFiUDP ntpUDP;
//NTP time
#include <NTPClient.h>
NTPClient timeClient(ntpUDP, "pool.ntp.org", 25200); //VN UTC timeset of VN = GMT+7 * 60 * 60
// WIFI NETWORK
const char* ssid = "abc";
const char* password = "abc";
// Set Static IP address
//IPAddress local_IP(192, 168, 0, 250);
//// Set other for IP address
//IPAddress gateway(192, 168, 0, 1);
//IPAddress subnet(255, 255, 255, 0);
//IPAddress primaryDNS(8, 8, 8, 8);   //optional
//IPAddress secondaryDNS(8, 8, 4, 4); //optional
[... 163 lines cut ...]
> 
> OK
> AT
> 
> OK
> AT
> 
> OK
> 
> SMS DONE
> 
> +SMS FULL
> 
> +CGEV: EPS PDN ACT 1
> 
> PB DONE
> ATD+abc;
> 
> +CLCC: 1,0,2,0,0,"+abc",145
> 
> OK

It said call okay but not, and 02 leds turns off (I changed my own info).

Update you for advise!

SeeNguyen, 2023-12-05:
HI,
So any advise?

lewisxhe, 2023-12-07:
anymore question ? Can you explain it?

SeeNguyen, 2023-12-07:
> anymore question ? Can you explain it?

question: can you make this run?

what do you want me to explain? all are above

lewisxhe, 2023-12-07:
I think the above output log ATD command should be followed by the mobile phone number. Did you deliberately change it?

SeeNguyen, 2023-12-07:
no value