Xinyuan-LilyGO/LilyGo-Modem-Series #73: A7670G GPS AT+CGNSSPWR=1 ERROR

sylwesterm, 2023-11-24:
I bought A7670G with attached GPS antenna. I tried several different examples but when call command AT+CGNSSPWR=1 it returns ERROR. I read somewhere that this device has no built in GPS. I tried two examples such GPSShield and GPSBuiltin. If this is the case that the A7670G doesn't have a GPS why the board has GNNS connector and attached GPS antenna??

lewisxhe, 2023-11-24:
7670G does not have GPS function

sylwesterm, 2023-11-24:
I ordered this part (see attached picture) It says it has GPS.
[image]
Why this board has GNSS connector and why it sold with GPS antenna??

lewisxhe, 2023-11-25:
The GPs connector on the front of the motherboard is invalid, and the Gps module is located on the back of the board, next to the 18650 battery holder

sylwesterm, 2023-11-27:
Can you show it?
[image]

lewisxhe, 2023-11-28:
See there https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/56#issuecomment-1672628977

closed 2024-01-12 by github-actions[bot]