Xinyuan-LilyGO/LilyGo-Modem-Series #75: A7670G SSL EC formats

G7OCD, 2023-12-29:
It works with HTTPS server with RSA 2048bit public key and Elliptic Curve 256bit public key.

BUT, If I connect to a server with EC 384bit then I get HTTP error 751 (TLS handshake failure).

Is there a new modem firmware which supports contemporary TLS suite? My modem has firmare V1.11.1

lewisxhe, 2024-01-05:
There is currently no firmware that supports EC formart

G7OCD, 2024-01-05:
I don't think that is quite correct. EC256 keyed certificates are processed OK, just not EC384.

Can you confirm modem v1.11.1 is the latest?

lewisxhe, 2024-01-09:
V1.11.1?
This is not the modem firmware version. Send AT+SIMCOMATI to check the return result.

G7OCD, 2024-01-10:
Because ATI and AT+SIMCOMATI provide different revision numbers......

ATI
Manufacturer: INCORPORATED
Model: A7670G-LLSE
Revision: A7670M7_V1.11.1

AT+SIMCOMATI
Manufacturer: INCORPORATED
Model: A7670G-LLSE
Revision: A011B17A7670M7
A7670M7_B17V01_230621

lewisxhe, 2024-01-11:
I need to check with SIMCOM

lewisxhe, 2024-01-12:
SIMCOM replied that the following are supported
```
0 SSL3.0
1 TLS1.0
2 TLS1.1
3 TLS1.2
```
No EC384

closed 2024-02-26 by github-actions[bot]