Xinyuan-LilyGO/LilyGo-Modem-Series #76: LilyGO-T-A7670G

LB3000, 2024-01-04:
Hello everyone,

I'm a beginner. I bought this board (LilyGO-T-A7670G) and I want to use it with Arduino IDE.

I've installed the libraries but I can't communicate with the board.

Which Arduino IDE board should I choose to use this board?

Why when I try some boards like ESP32 dev module to see if it can work, the download doesn't make my board flash. The blue LED lights up continuously, as well as two red LEDs.

Do you have a code to try to make the blue led blink to see if my upload and communication with the card is working?

I'm a real beginner. I've already managed to get a simple ESP32 to work, but I'm stuck with this card. I'm also open to any resource to help me (link to projects with this board, starting a connected object project, etc).

Thank you,
Best regards

daveymg-nz, 2024-01-04:
The Lilygo libraries allow you to setup and manage the comms to the modem. 

You could try doing a simple print Hello World sketch to confirm that your upload is working ok. 

The libraries have example code to check the modem and allow you tosend it commands from the keyboard.

lewisxhe, 2024-01-05:
Please check the homepage description

LB3000, 2024-01-05:
Hello,

Thank you for your answers,

As a beginner, I didn't even know how to make a "Hello world". While looking for how to do it, I discovered that I wasn't looking at my output monitor on Arduino IDE, the board responds fine. That's how I found my error.

For the question I was asking myself, in the description page, it doesn't say which board to choose in Arduino IDE. By choosing the "ESP32 Dev Module" board model, you can communicate with a LILYGO board. This information seems to be missing. 

I had read the resources, but it was a simple beginner's error in the arduino IDE software.

Thank you

daveymg-nz, 2024-01-05:
As a beginner you may be better off getting an esp8266 and the sensors you wish to use and experiment with coding on that rather than risk damaging the lilygo board before you have built up some coding experience.

LB3000, 2024-01-05:
Yes, I'm trying to code my sensors on a simple ESP32 now before going back to the board.
But I have no choice for my project, I need to integrate an LTE connection, which this board offers. With the explanations in the examples, I shouldn't make too many mistakes. I hope aha

daveymg-nz, 2024-01-05:
If you get all of your data transfers working correctly using WiFi to begin with, it's fairly straight forward to switch it to using the LTE modem. If you're using a lot of sensors you need to be careful with the gpio pins that you use on the lilygo board as a number of them are already used for other purposes.

closed 2024-02-20 by github-actions[bot]