Xinyuan-LilyGO/LilyGo-Modem-Series #77: Error +CMQTTCONNECT: 0,34 while connecting to broker.emqx.io:8883

mranest, 2024-01-08:
Following the documentation, and also relevant discussion found in https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/49. While I am able to connect to plain, non-TLS brokers, trying both `broker.emqx.io` and hivemq private broker on port 8883 I get error `+CMQTTCONNECT: 0,34`. Steps are as follows:

```
AT+CMQTTSTART
OK

+CMQTTSTART: 0
AT+CCERTDOWN="ca_cert.pem",1338
> [...]
OK
AT+CSSLCFG="sslversion",0,4
OK
AT+CSSLCFG="cacert",0,"ca_cert.pem"
OK
AT+CSSLCFG="enableSNI",0,1
OK
AT+CSSLCFG="authmode",0,1
OK
AT+CMQTTSSLCFG=0,0
OK
AT+CMQTTACCQ=0,"A7670",1
OK
AT+CMQTTCONNECT=0,"tcp://broker.emqx.io:8883",60,1
+CMQTTCONNECT: 0,34

ERROR
```

Output of `AT+SIMCOMATI` is as follows:

```
Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B04A7670M7_F
A7670M7_B04V02_220927
QCN:
IMEI: XXX
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:
```

Please advise.

G7OCD, 2024-01-10:
The cert you are uploading isn't the CA for that site. openssl s_client reports
Digicert as the correct CA. The server cert is issued to a wildcard... not sure 
if there may be some strange interaction between that and SNI in this
device.

mranest, 2024-01-11:
This is the CA certificate that is offered in the site itself. And it is indeed the Digicert one. Checking with openssl I get:

```
❯ openssl x509 -in broker.emqx.io-ca.pem -text
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            03:3a:f1:e6:a7:11:a9:a0:bb:28:64:b1:1d:09:fa:e5
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: C=US, O=DigiCert Inc, OU=www.digicert.com, CN=DigiCert Global Root G2
[...]
```

I understand what you say about the wildcard, but this is something beyond my control. Is it a known issue with the SIMCOM module?

lewisxhe, 2024-01-11:
```

OK
AT+CSQ
AT+CSQ
+CSQ: 27,99

OK
[17] Signal Quality:27AT+CGREG?

+CGEV: EPS PDN ACT 1
AT+CGREG?
+CGREG: 0,1

OK
Online registration successful

Registration Status:1
AT+CPSI?
AT+CPSI?
+CPSI: LTE,Online,460-11,0x775C,117004565,333,EUTRAN-BAND1,100,5,0,20,58,56,17

OK
Inquiring UE system information:LTE,Online,460-11,0x775C,117004565,333,EUTRAN-BAND1,100,5,0,20,58,56,17
AT+NETOPEN
AT+NETOPEN
OK

+NETOPEN: 0AT+IPADDR

SMS DONE
AT+IPADDR
+IPADDR: 10.14.122.17

OK
Network IP:10.14.122.17
AT+CMQTTSTART
AT+CMQTTSTART
OK

+CMQTTSTART: 0
Connecting to broker.emqx.ioAT+CCERTDOWN="ca_cert.pem",1295
AT+CCERTDOWN="ca_cert.pem",1295
>
-----BEGIN CERTIFICATE-----
MIIDjjCCAnagAwIBAgIQAzrx5qcRqaC7KGSxHQn65TANBgkqhkiG9w0BAQsFADBh
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3
d3cuZGlnaWNlcnQuY29tMSAwHgYDVQQDExdEaWdpQ2VydCBHbG9iYWwgUm9vdCBH
MjAeFw0xMzA4MDExMjAwMDBaFw0zODAxMTUxMjAwMDBaMGExCzAJBgNVBAYTAlVT
MRUwEwYDVQQKEwxEaWdpQ2VydCBJbmMxGTAXBgNVBAsTEHd3dy5kaWdpY2VydC5j
b20xIDAeBgNVBAMTF0RpZ2lDZXJ0IEdsb2JhbCBSb290IEcyMIIBIjANBgkqhkiG
9w0BAQEFAAOCAQ8AMIIBCgKCAQEAuzfNNNx7a8myaJCtSnX/RrohCgiN9RlUyfuI
2/Ou8jqJkTx65qsGGmvPrC3oXgkkRLpimn7Wo6h+4FR1IAWsULecYxpsMNzaHxmx
1x7e/dfgy5SDN67sH0NO3Xss0r0upS/kqbitOtSZpLYl6ZtrAGCSYP9PIUkY92eQ
q2EGnI/yuum06ZIya7XzV+hdG82MHauVBJVJ8zUtluNJbd134/tJS7SsVQepj5Wz
tCO7TG1F8PapspUwtP1MVYwnSlcUfIKdzXOS0xZKBgyMUNGPHgm+F6HmIcr9g+UQ
vIOlCsRnKPZzFBQ9RnbDhxSJITRNrw9FDKZJobq7nMWxM4MphQIDAQABo0IwQDAP
BgNVHRMBAf8EBTADAQH/MA4GA1UdDwEB/wQEAwIBhjAdBgNVHQ4EFgQUTiJUIBiV
5uNu5g/6+rkS7QYXjzkwDQYJKoZIhvcNAQELBQADggEBAGBnKJRvDkhj6zHd6mcY
1Yl9PMWLSn/pvtsrF9+wX3N3KjITOYFnQoQj8kVnNeyIv/iPsGEMNKSuIEyExtv4
NeF22d+mQrvHRAiGfzZ0JFrabA0UWTW98kndth/Jsw1HKj2ZL7tcu7XUIOGZX1NG
Fdtom/DzMNU+MeKNhJ7jitralj41E6Vf8PlwUHBHQRFXGU7Aj64GxJUTFy8bJZ91
8rGOmaFvE7FBcf6IKshPECBV1/MUReXgRPTqh5Uykw7+U0b6LJ3/iyK5S9kJRaTe
pLiaWN0bfVKfjllDiIGknibVb63dDcY3fe0Dkhvld1927jyNxF1WW6LZZm6zNTfl
MrY=
-----END CERTIFICATE-----

OK
AT+CSSLCFG="cacert",0,"ca_cert.pem"
AT+CSSLCFG="cacert",0,"ca_cert.pem"
OK
AT+CSSLCFG="sslversion",0,4
AT+CSSLCFG="sslversion",0,4
OK
AT+CMQTTSSLCFG=0,0
AT+CMQTTSSLCFG=0,0
OK
AT+CSSLCFG="authmode",0,1
AT+CSSLCFG="authmode",0,1
OK
AT+CMQTTREL=0
AT+CMQTTREL=0
+CMQTTREL: 0,20

ERROR
AT+CMQTTACCQ=0,"A76XX",1
AT+CMQTTACCQ=0,"A76XX",1
OK
AT+CMQTTCONNECT=0,"tcp://broker.emqx.io:8883",60,1
AT+CMQTTCONNECT=0,"tcp://broker.emqx.io:8883",60,1
OK

+CMQTTCONNECT: 0,0successed.
AT+CMQTTDISC?

AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK
MQTT has connected!
AT+CMQTTSUBTOPIC=0,21,0
AT+CMQTTSUBTOPIC=0,21,0
>GsmMqttTest/subscribe

OK
AT+CMQTTSUB=0
AT+CMQTTSUB=0
OK

+CMQTTSUB: 0,0
AT+CMQTTTOPIC=0,19
AT+CMQTTTOPIC=0,19
>GsmMqttTest/publish

OK
AT+CMQTTPAYLOAD=0,10
AT+CMQTTPAYLOAD=0,10
>RunTime:27

OK
AT+CMQTTPUB=0,0,60
AT+CMQTTPUB=0,0,60
OK

+CMQTTPUB: 0,0
AT+CMQTTDISC?
AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK
AT+CMQTTTOPIC=0,19
AT+CMQTTTOPIC=0,19
>GsmMqttTest/publish

OK
AT+CMQTTPAYLOAD=0,10
AT+CMQTTPAYLOAD=0,10
>RunTime:37

OK
AT+CMQTTPUB=0,0,60
AT+CMQTTPUB=0,0,60
OK

+CMQTTPUB: 0,0

PB DONE
AT+CMQTTDISC?
AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK
AT+CMQTTTOPIC=0,19
AT+CMQTTTOPIC=0,19
>GsmMqttTest/publish

OK
AT+CMQTTPAYLOAD=0,10
AT+CMQTTPAYLOAD=0,10
>RunTime:47

OK
AT+CMQTTPUB=0,0,60
AT+CMQTTPUB=0,0,60
OK

+CMQTTPUB: 0,0
AT+CMQTTDISC?
AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK
AT+CMQTTTOPIC=0,19
AT+CMQTTTOPIC=0,19
>GsmMqttTest/publish

OK
AT+CMQTTPAYLOAD=0,10
AT+CMQTTPAYLOAD=0,10
>RunTime:57

OK
AT+CMQTTPUB=0,0,60
AT+CMQTTPUB=0,0,60
OK

+CMQTTPUB: 0,0
AT+CMQTTDISC?
AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK
AT+CMQTTTOPIC=0,19
AT+CMQTTTOPIC=0,19
>GsmMqttTest/publish

OK
AT+CMQTTPAYLOAD=0,10
AT+CMQTTPAYLOAD=0,10
>RunTime:67

OK
AT+CMQTTPUB=0,0,60
AT+CMQTTPUB=0,0,60
OK

+CMQTTPUB: 0,0
```

lewisxhe, 2024-01-11:
Why do you want to enable SNI? ? EMQX does not need to enable SNI

mranest, 2024-01-11:
Found the enableSNI in a sample I used, and I think it is required for the private brokers in Emqx. Nevertheless I tried the same commands as @lewisxhe suggested:

```
*ATREADY: 1

+CPIN: READY

SMS DONE

+CGEV: EPS PDN ACT 1

PB DONE
AT+CSQ
+CSQ: 16,99

OK
AT+CGREG?
+CGREG: 0,5

OK
AT+CPSI?
+CPSI: LTE,Online,202-05,0xBD3,2597675,30,EUTRAN-BAND1,100,5,0,0,31,38,3

OK
AT+NETOPEN
OK

+NETOPEN: 0
AT+IPADDR
+IPADDR: 10.22.22.135

OK
AT+CMQTTSTART
OK

+CMQTTSTART: 0
AT+CCERTDOWN="ca_cert.pem",1295
>
OK
AT+CSSLCFG="cacert",0,"ca_cert.pem"
OK
AT+CSSLCFG="sslversion",0,4
OK
AT+CMQTTSSLCFG=0,0
OK
AT+CSSLCFG="authmode",0,1
OK
AT+CMQTTREL=0
+CMQTTREL: 0,20

ERROR
AT+CMQTTACCQ=0,"A76XX",1
OK
AT+CMQTTCONNECT=0,"tcp://broker.emqx.io:8883",60,1
+CMQTTCONNECT: 0,34

ERROR
```

Two differences I see. One is I use `A7670E-FASE` instead of `A7670SA-FASE`, and the other is my SIM is an IoT one that is using roaming through other providers (hence the `+CGREG: 0,5` difference in response).

lewisxhe, 2024-01-11:
A7670SA-FASE and A7670E only differ in the radio frequency part, and the internal firmware is exactly the same.

mranest, 2024-01-11:
Agreed. So that leaves the roaming as a potential issue?

lewisxhe, 2024-01-11:
[image]

Check the returned results. It seems more like there is no network access. You may need to configure APN. In some countries, SIM cards require manual configuration of APN. In my country, it is not required.

lewisxhe, 2024-01-11:
Place this code before turning on the network
```
const char *apn = "Your APN";
    //Set the APN manually. Some operators need to set APN first when registering the network.
    modem.sendAT("+CGDCONT=1,\"IP\",\"", apn, "\"");
    if (modem.waitResponse() != 1) {
        Serial.println("Set operators apn Failed!");
        return;
    }
```

mranest, 2024-01-11:
Apologies, I just saw the issue description and I haven't described the workaround correctly.

If I disable server authentication (i.e., `"authmode",0,0`) I can connect to the broker without problems. So continuing the same session as left above, after the `+CMQTTCONNECT: 0,34` error, if I just give:

```
AT+CSSLCFG="authmode",0,0
OK
AT+CMQTTCONNECT=0,"tcp://broker.emqx.io:8883",60,1
OK

+CMQTTCONNECT: 0,0
```

My SIM also does not need explicit configuration of the APN, as soon as it becomes ready it automatically connects to the network (hence the `+CGEV: EPS PDN ACT 1` unsolicited message in the very beginning).

So, definitely has something to do with certificate validation, but can't understand how in your case it works without issues.

lewisxhe, 2024-01-11:
I noticed that your certificate name is broker.emqx.io-ca.pem. Please change the certificate name to ca_cert.pem. I have updated the example. You can use the example I provided for testing.

mranest, 2024-01-11:
Ran it again with ca_cert.pem, still no go:

```
AT+CMQTTSTART
OK

+CMQTTSTART: 0
AT+CCERTDOWN="ca_cert.pem",1295
>
OK
AT+CSSLCFG="cacert",0,"ca_cert.pem"
OK
AT+CSSLCFG="sslversion",0,4
OK
AT+CMQTTSSLCFG=0,0
OK
AT+CSSLCFG="authmode",0,1
OK
AT+CMQTTREL=0
+CMQTTREL: 0,20

ERROR
AT+CMQTTCONNECT=0,"tcp://broker.emqx.io:8883",60,1
+CMQTTCONNECT: 0,20

ERROR
AT+CMQTTACCQ=0,"A76XX",1
OK
AT+CMQTTCONNECT=0,"tcp://broker.emqx.io:8883",60,1
+CMQTTCONNECT: 0,34

ERROR
```

lewisxhe, 2024-01-11:
emmm.. Restore factory default parameters and try again,
```
AT&F
```
Due to the closed nature of SIMCOM, we cannot obtain more meaningful data from the return value. I am helpless when facing this problem.

mranest, 2024-01-11:
Thank you for the help though @lewisxhe, I understand the constraints.

One more thing if I may. Can you run the following set of commands after having uploaded the certificate, set the SSLCFG etc, just to compare with my side?

```
AT+CCHSTART
OK

+CCHSTART: 0
AT+CCHSSLCFG=0,0
OK
AT+CCHOPEN=0,"broker.emqx.io",8883,2
OK

+CCHOPEN: 0,15
```

I'm following the instructions found in `3.2 Access to SSL/TLS server (not verify server and client)`, in `A76XX Series_SSL_Application Note_V1.02.pdf`. Error 15 is a `Handshake error`.

mranest, 2024-01-11:
I am also failing to connect to HTTPS when attempting to validate CA certificate. E.g.:

```
AT+CCHOPEN=0,"cloud-intl.emqx.com",443,2
```

So either there is a problem with the module while doing the handshake, or there is a Man-In-The-Middle situation with my SIM that is intercepting and changing TLS and there is a legit validation issue.

Do you know if there is any way to get some debug information out of the CA validation process from the module?

G7OCD, 2024-01-11:
There could be a number of reasons that TLS fails but plain works:
- Server and SIMCOM module don't have a common cypher suite or key handling.
- The module is not setting the clock (check out AT CTZR and CCLK).
- Your mobile carrier is blocking TLS for some reason (local laws etc?).

For me, I first had to use CTZR to get the clock to take network time and
then I had to diganose lack os support for 384bit EC keys in the module.

G7OCD, 2024-01-11:
Using s_client to check emqx.com it looks like RSA2048 so it shouldn't be that.

mranest, 2024-01-11:
> There could be a number of reasons that TLS fails but plain works:
> 
> * Server and SIMCOM module don't have a common cypher suite or key handling.
> * The module is not setting the clock (check out AT CTZR and CCLK).
> * Your mobile carrier is blocking TLS for some reason (local laws etc?).
> 
> For me, I first had to use CTZR to get the clock to take network time and then I had to diganose lack of support for 384bit EC keys in the module.

Thank you for the hints! I will check the CTZR and CCLK. I am pretty sure it's properly synchronising with the network, as I use it to sync the ESP32 RTC but I am not sure about the TZ.

I think though that if TLS handshake was failing due to clock discrepancy it would also fail when setting authmode to 0,0 (i.e., not attempting to validate server). I can successfully connect to the TLS port of the broker this way.

lewisxhe, 2024-01-12:
> Thank you for the help though @lewisxhe, I understand the constraints.
>
> One more thing if I may. Can you run the following set of commands after having uploaded the certificate, set the SSLCFG etc, [...]

[image]

lewisxhe, 2024-01-12:
```
OK
AT+CSQ
AT+CSQ
+CSQ: 27,99

OK
[16] Signal Quality:27AT+CGREG?

+CGEV: EPS PDN ACT 1
AT+CGREG?
+CGREG: 0,1

OK
Online registration successful

Registration Status:1
AT+CPSI?
AT+CPSI?
+CPSI: LTE,Online,460-11,0x775C,125414932,241,EUTRAN-BAND5,2452,3,0,19,53,52,3

OK
Inquiring UE system information:LTE,Online,460-11,0x775C,125414932,241,EUTRAN-BAND5,2452,3,0,19,53,52,3
AT+NETOPEN
AT+NETOPEN
OK

+NETOPEN: 0AT+IPADDR

SMS DONE
AT+IPADDR
+IPADDR: 10.17.51.118

OK
Network IP:10.17.51.118
AT+CMQTTSTART
AT+CMQTTSTART
OK

+CMQTTSTART: 0
Connecting to broker.emqx.ioAT+CCERTDOWN="ca_cert.pem",1295
AT+CCERTDOWN="ca_cert.pem",1295
>
-----BEGIN CERTIFICATE-----
MIIDjjCCAnagAwIBAgIQAzrx5qcRqaC7KGSxHQn65TANBgkqhkiG9w0BAQsFADBh
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3
d3cuZGlnaWNlcnQuY29tMSAwHgYDVQQDExdEaWdpQ2VydCBHbG9iYWwgUm9vdCBH
MjAeFw0xMzA4MDExMjAwMDBaFw0zODAxMTUxMjAwMDBaMGExCzAJBgNVBAYTAlVT
MRUwEwYDVQQKEwxEaWdpQ2VydCBJbmMxGTAXBgNVBAsTEHd3dy5kaWdpY2VydC5j
b20xIDAeBgNVBAMTF0RpZ2lDZXJ0IEdsb2JhbCBSb290IEcyMIIBIjANBgkqhkiG
9w0BAQEFAAOCAQ8AMIIBCgKCAQEAuzfNNNx7a8myaJCtSnX/RrohCgiN9RlUyfuI
2/Ou8jqJkTx65qsGGmvPrC3oXgkkRLpimn7Wo6h+4FR1IAWsULecYxpsMNzaHxmx
1x7e/dfgy5SDN67sH0NO3Xss0r0upS/kqbitOtSZpLYl6ZtrAGCSYP9PIUkY92eQ
q2EGnI/yuum06ZIya7XzV+hdG82MHauVBJVJ8zUtluNJbd134/tJS7SsVQepj5Wz
tCO7TG1F8PapspUwtP1MVYwnSlcUfIKdzXOS0xZKBgyMUNGPHgm+F6HmIcr9g+UQ
vIOlCsRnKPZzFBQ9RnbDhxSJITRNrw9FDKZJobq7nMWxM4MphQIDAQABo0IwQDAP
BgNVHRMBAf8EBTADAQH/MA4GA1UdDwEB/wQEAwIBhjAdBgNVHQ4EFgQUTiJUIBiV
5uNu5g/6+rkS7QYXjzkwDQYJKoZIhvcNAQELBQADggEBAGBnKJRvDkhj6zHd6mcY
1Yl9PMWLSn/pvtsrF9+wX3N3KjITOYFnQoQj8kVnNeyIv/iPsGEMNKSuIEyExtv4
NeF22d+mQrvHRAiGfzZ0JFrabA0UWTW98kndth/Jsw1HKj2ZL7tcu7XUIOGZX1NG
Fdtom/DzMNU+MeKNhJ7jitralj41E6Vf8PlwUHBHQRFXGU7Aj64GxJUTFy8bJZ91
8rGOmaFvE7FBcf6IKshPECBV1/MUReXgRPTqh5Uykw7+U0b6LJ3/iyK5S9kJRaTe
pLiaWN0bfVKfjllDiIGknibVb63dDcY3fe0Dkhvld1927jyNxF1WW6LZZm6zNTfl
MrY=
-----END CERTIFICATE-----

OK
AT+CSSLCFG="cacert",0,"ca_cert.pem"
AT+CSSLCFG="cacert",0,"ca_cert.pem"
OK
AT+CSSLCFG="sslversion",0,4
AT+CSSLCFG="sslversion",0,4
OK
AT+CMQTTSSLCFG=0,0
AT+CMQTTSSLCFG=0,0
OK
AT+CSSLCFG="authmode",0,1
AT+CSSLCFG="authmode",0,1
OK
AT+CMQTTREL=0
AT+CMQTTREL=0
+CMQTTREL: 0,20

ERROR
AT+CMQTTACCQ=0,"A76XX",1
AT+CMQTTACCQ=0,"A76XX",1
OK
AT+CMQTTCONNECT=0,"tcp://broker.emqx.io:8883",60,1
AT+CMQTTCONNECT=0,"tcp://broker.emqx.io:8883",60,1
OK

+CMQTTCONNECT: 0,0successed.
AT+CMQTTDISC?

AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK
MQTT has connected!
AT+CMQTTSUBTOPIC=0,21,0
AT+CMQTTSUBTOPIC=0,21,0
>GsmMqttTest/subscribe

OK
AT+CMQTTSUB=0
AT+CMQTTSUB=0
OK

+CMQTTSUB: 0,0
AT+CMQTTDISC=0,120
AT+CMQTTDISC=0,120
OK

+CMQTTDISC: 0,0
AT+CMQTTREL=0
AT+CMQTTREL=0
OK
AT+CMQTTSTOP
AT+CMQTTSTOP
OK

+CMQTTSTOP: 0

PB DONE
AT+SIMCOMATI
Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B04A7670M7_F
A7670M7_B04V02_220927
QCN:
IMEI: 860470xxxxxx
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK
AT+CCHSTART
OK

+CCHSTART: 0
AT+CCHSSLCFG=0,0
OK
AT+CCHOPEN=0,"broker.emqx.io",8883,2
OK

+CCHOPEN: 0,0
```
I did the test again using A7670E

mranest, 2024-01-12:
Thank you @lewisxhe. Definitely something fishy on my part. 

Guess there is no more debugging that can be retrieved from the module (i.e., something resembling openssl s_client)?

lewisxhe, 2024-01-12:
As I mentioned above, SIMCOM's firmware is too closed and can return very little effective information. If something goes wrong, it is particularly difficult to determine where the problem is.
emqx is from China, I'm not sure if it's because your area is blocked by the firewall, or for other reasons, can you try a local MQTT server in your area for testing?

mranest, 2024-01-14:
So I verified CCLK was correctly set, and also corrected the TZ. Still getting the same error. As established this seems to be some problem on my part, so feel free to close this issue. Thank you @lewisxhe  and @G7OCD for your help and hints.

closed 2024-02-29 by github-actions[bot]