Xinyuan-LilyGO/LilyGo-Modem-Series #79: A7670G +CME error sim failure

sharathk, 2024-01-17:
Hi,

I have two board of A7670G and both gives me 
+CME ERROR sim failure
and can't connect to network.
But I tried the sim card on a TTGO-7000G I had and it works.
Do you know what might be the problem.

Thanks

lewisxhe, 2024-01-18:
This error was returned after sending what command. Is there a complete process?

ruben148, 2024-02-10:
Hello! Have you solved this? I have the same issue. I'm trying to run the HiveMQ mqtt example (modified a bit) but it looks like it doesn't want to connect to the network. I'm having a similar problem with making a call or sending sms. The module can however obtain the SMSC number which I think is weird. I tried with multiple SIM cards which all work on other IoT devices.
I live in Romania and the 4G bands for my carrier are 3/20 which should of course be supported by the A7670G.
Another weird thing is that the modem gets denied several times when trying to connect. In the original code, if the modem gets denied, the program stops.

Here's the code I'm trying to run:
```
void setup()
{
    Serial.begin(115200);

    Serial.println("Start Sketch");

    SerialAT.begin(115200, SERIAL_8N1, MODEM_RX_PIN, MODEM_TX_PIN);

#ifdef BOARD_POWERON_PIN
    pinMode(BOARD_POWERON_PIN, OUTPUT);
    digitalWrite(BOARD_POWERON_PIN, HIGH);
#endif

    // Set modem reset pin ,reset modem
    pinMode(MODEM_RESET_PIN, OUTPUT);
    digitalWrite(MODEM_RESET_PIN, !MODEM_RESET_LEVEL); delay(100);
    digitalWrite(MODEM_RESET_PIN, MODEM_RESET_LEVEL); delay(2600);
    digitalWrite(MODEM_RESET_PIN, !MODEM_RESET_LEVEL);

    pinMode(BOARD_PWRKEY_PIN, OUTPUT);
    digitalWrite(BOARD_PWRKEY_PIN, LOW);
    delay(100);
    digitalWrite(BOARD_PWRKEY_PIN, HIGH);
    delay(100);
    digitalWrite(BOARD_PWRKEY_PIN, LOW);

    // Check if the modem is online
    Serial.println("Start modem...");

    int retry = 0;
    while (!modem.testAT(1000)) {
        Serial.println(".");
        if (retry++ > 10) {
            digitalWrite(BOARD_PWRKEY_PIN, LOW);
            delay(100);
            digitalWrite(BOARD_PWRKEY_PIN, HIGH);
            delay(1000);
            digitalWrite(BOARD_PWRKEY_PIN, LOW);
            retry = 0;
        }
        delay(1000);
    }

    SimStatus sim = SIM_ERROR;

    while (sim != SIM_READY) {
        sim = modem.getSimStatus();
        switch (sim) {
        case SIM_READY:
            Serial.println("SIM card online.");
            break;
        case SIM_LOCKED:
            Serial.println("The SIM card is locked. Unlocking...");
            //modem.simUnlock(SIMCARD_PIN_CODE);
            break;
        default:
            break;
        }
        delay(1000);
    }
    delay(10000);
    
    //SIM7672G Can't set network mode
#ifndef TINY_GSM_MODEM_SIM7672
[... 381 lines cut ...]
[75] Signal Quality:23AT+CPSI?
AT+CPSI?
+CPSI: GSM,Online,226-10,0x206C,21259,84 PGSM 900,45,0,43-43

OK
Inquiring UE system information:GSM,Online,226-10,0x206C,21259,84 PGSM 900,45,0,43-43
AT+NETOPEN
AT+NETOPEN
OK

+CGEV: ME PDN DEACT 1
Enable network failed!
AT+IPADDR
AT+IPADDR
+IP ERROR: Network not opened

ERROR
Network IP:
AT+CMQTTSTART
AT+CMQTTSTART
OK

+CMQTTSTART: 0
...
```

closed 2024-03-31 by github-actions[bot]

deeja, 2025-05-19:
Had the same issue. Put a new sim card in and it worked. So it was either the connection or the sim card.