Xinyuan-LilyGO/LilyGo-Modem-Series #80: A7670SA-FASE NMEA Parsing error TinyGPSPlus

AlverGant, 2024-01-19:
Hi,

Please help.

I have the following modem:
Manufacturer: INCORPORATED
Model: A7670SA-FASE
Revision: A011B04A7670M7_F
A7670M7_B04V02_220927
QCN:
IMEI: 869731059073809
MEID:
+GCAP: +CGSM,+FCLASS,+DS

I am trying to parse GPS data and it appears the modem is spilling a correct NMEA but parsing with https://github.com/mikalhart/TinyGPSPlus is giving me wrong coordinates...

Here is my startup GPS code 

modem.sendAT("+CGDRT=4,1");
modem.sendAT("+CGSETV=4,0");
modem.sendAT("+CGNSSPWR=1,1");
modem.sendAT("+CGPS=0");
modem.sendAT("+CGNSSMODE=3");
modem.sendAT("+CGPSNMEA=1,1,1,1,1,1,0,0");
modem.sendAT("+CGPSNMEARATE=2");
modem.sendAT("+CGNSSTST=1");
modem.sendAT("+CGNSSPORTSWITCH=0,1");

AT+CGNSSINFO
+CGNSSINFO: 3,15,,05,07,22.9284973,S,43.1748390,W,190124,064019.00,110.3,2.196,,18.29,10.06,15.28,
OK

But parsing with modem.getGPS(&lat, &lon)
gives me wrong coordinates
Latitude: -0.116667, Longitude: -0.000000

What could be wrong here

lewisxhe, 2024-01-19:
A011B04A7670M7_F This version of GPS should not have problems. I checked that the location you provided should be correct. Please try to use the NMEA Parse example I provided for testing.
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/GPS_NMEA_Parse

AlverGant, 2024-01-19:
Thank you @lewisxhe for the prompt return

I am trying to compile your code having added:
#define LILYGO_T_A7670
#define TINY_GSM_MODEM_SIM7600

But I am encountering issues with the gps library:
Compilation error: 'TinyGsm' {aka 'class TinyGsmSim7600'} has no member named 'setGPSBaud'; did you mean 'setBaud'?

Please what gps library are you using?

Best regards

lewisxhe, 2024-01-19:
Are you using the previous example? Please see README and reinstall TinyGSM in the warehouse. Do not use upstream TinyGSM. Upstream TinyGSM does not support A7670.

AlverGant, 2024-01-19:
Thank you again @lewisxhe that was my problem indeed.

Your code is working!

Will integrate it into my code

lewisxhe, 2024-01-19:
Um . I added many useful examples that you can test and if the problem is solved please close it

closed 2024-01-19 by AlverGant