Xinyuan-LilyGO/LilyGo-Modem-Series #81: analogRead when battery powered

GabrielZre, 2024-02-08:
Hi, when I powered A7608E-H from Usb C cable I got an accurate values on 3.55, but when switched into battery source (18650 3.7V), then the values are not even close. 
How should I get this value? Is it constant? I don't think so if the battery source really makes difference.

```
rawVoltageValue = analogRead(voltagePin);
voltageValueArduino = (rawVoltageValue * 3.55) / 4095.0;
```

lewisxhe, 2024-03-05:
VOLTAGEPIN = 0? Where did you know that ADC Pin is 0?
There are two versions of A7608, one version has a battery voltage divider circuit (using GPIO35), and the other one has, which version do you use?

GabrielZre, 2024-03-05:
A7608E-H, actually I compared few pins and changed this pin 0 to 35, so now
I am using pin 34 and 35, the values are equals now

lewisxhe, 2024-03-06:
A7608 uses GPIO35. You can check the schematic diagram. The voltage dividing resistor is connected to GPIO35.

closed 2024-04-20 by github-actions[bot]