Xinyuan-LilyGO/LilyGo-Modem-Series #82: Telegram bot with TLS 1.2 & TinyGsm & Universal Telegram Bot - collaboration?

Rivelino911, 2024-02-11:
Is not an Issue, is an Enhancement.

As Telegram don't work anymore with SIM800L, can you talk with TinyGSM and Universal Telegram bot to make an integration?

lewisxhe, 2024-03-05:
No plans at the moment, may have to rely on others

But you can send me a link to see what it is, I haven't heard of it.

Rivelino911, 2024-03-05:
Good evening  Mr,
They create library  for serial communication with Simcom. Modules. As your LilyGo are as an compact product, is very good one for different applications. PS: In future can you add Ring Indicator tied to a thruhole somewhere. What i saw in your diagram RI is not connected on sim7000 also not in sim800. https://github.com/vshymanskyy/TinyGSM

lewisxhe, 2024-03-06:
> PS: In future can you add Ring Indicator tied to a thruhole somewhere. 
What does it mean ?
> What i saw in your diagram RI is not connected on sim7000 also not in sim800. 
I personally do not recommend SIM800 and SIM7000. SIM800 is no longer available in my country because we do not have 2G here. SIM7000 is unless you use NBIOT. If you use 2G, I recommend A7670. A7670 is easier to use than SIM800 and SIM7000.
> https://github.com/vshymanskyy/TinyGSM
vhymanskyy has no maintenance on TinyGSM, so what are you referring to?

closed 2024-04-20 by github-actions[bot]