Xinyuan-LilyGO/LilyGo-Modem-Series #84: T-A7670G bricked?

ruben148, 2024-02-20:
I have two T-A7670G. I used one of them a lot in the past few days. It worked perfectly fine until today when the modem started sending weird characters through Serial. When testing AT at the initialization step, it still sends multiple 'AT' but along with multiple weird characters. Then, when reaching the 'waitForNetwork' function, it cannot connect to the network (weird characters still present here). Finally, it gives me unhandled exception, kernel panic and reboots. Now I cannot even upload anything to the board. I would send the output here if I could see any communication between the board and the PC. The other board is working fine in the exact same conditions.

ruben148, 2024-02-20:
All right. I managed to upload a new sketch to it. Only sending a message through serial every second. Nothing modem related. It sends the message 6 times then stops. Then I cannot upload anything else even if the board is reset. Does this sound like a bricked board?

ruben148, 2024-03-05:
Yes, I cannot upload any sketch now. I tried the method in the FAQ using the buttons but also grounding the IO0 pin. It doesn't work either way.

lewisxhe, 2024-03-05:
Then please follow my steps to test it

1. Short-circuit IO0 to GND and keep connected
2. Use USB cable to connect the board and computer
3. Open the Serial Monitor, Arduino IDE Serial Monitor, or other, and adjust the baud rate to 115200
4. Press RST (reset) on the board

What do you see in the Serial monitor?

closed 2024-04-19 by github-actions[bot]