Xinyuan-LilyGO/LilyGo-Modem-Series #87: Firmware problem on A7608-E

emil151997, 2024-02-23:
Hello! 

I have two Lilygo A7608 boards. And there is a problem with connection (on one of them) - it is very unstable and reconnect often, while the other works much much better. 

Normal board:
```
AT+SIMCOMATI
Manufacturer: INCORPORATED
Model: A7608E-H
Revision: A50C4B07A7600M7
A7600M7_B07V01_220429
QCN:
IMEI: 860371050882459
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:
OK
```

Problem board:
```
AT+SIMCOMATI
Manufacturer: INCORPORATED
Model: A7608E-H
Revision: A50C4B08A7600M7
A7600M7_B08V02_220929
QCN:
IMEI: 861513060819158
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:
OK
```

So, it looks like the more recent firmware work much worse. So, can i get the most actual and non-problem firmware ?

lewisxhe, 2024-03-05:
Judging from the return, the problematic version is in the lead (B08), and the normal version is in (B07), right?

lewisxhe, 2024-03-05:
At present, it seems that the GPS positioning coordinates before the B11 version are wrong. You can try to update to the B11 version for testing.
https://drive.google.com/file/d/1IfNkPfQmfG3oqbXEZl0YD_9qgLsN4e_D/view?usp=sharing

emil151997, 2024-03-05:
To be honest, i found recent firmware (B11) in other issue. Bui it didn't help. So, it looks like the problem with hardware

lewisxhe, 2024-03-05:
What is the problem? Can you share it?

emil151997, 2024-04-13:
Sorry for the late answer, so the problem is, that controller after some time i start have problem with HTTP requests. error on HTTPAction commands. I have about 10 boards and all have strange problems - some of them lost connection, some have this 706 error. Can it be related to the hardware version of the board "XY-A4608 2022-7-18 V1.2" ? 

My application logic:
1. Modem hardware startup
`pinMode(MODEM_PWR_PIN, OUTPUT);
    pinMode(MODEM_RST_PIN, OUTPUT);

    digitalWrite(MODEM_RST_PIN, HIGH);
    delay(1000);

    digitalWrite(MODEM_PWR_PIN, LOW);
    delay(100);

    digitalWrite(MODEM_RST_PIN, HIGH);
    delay(1000);
    digitalWrite(MODEM_RST_PIN, LOW);

    digitalWrite(MODEM_PWR_PIN, LOW);
    delay(100);
    digitalWrite(MODEM_PWR_PIN, HIGH);
    delay(1000);
    digitalWrite(MODEM_PWR_PIN, LOW);
    delay(1000);

    Serial1.begin(MODEM_UART_BAUD, SERIAL_8N1, MODEM_RX_PIN, MODEM_TX_PIN);
    delay(1000);`
2. AT_CMD_INIT:
2.1 AT
2.2 ATE0
2.3 AT+CREG
2.4 AT+CEREG/AT+CGREG
2.5 If i have 3g or 2g (not LTE) - activate PDP Context with AT+CGAUTH, AT+CGDCONT, AT+CGACT commands 
3. Start Main Application - repeatedly HTTP POST/Get requests 
3.1 If i have problems with http request, i reset my controller with ESP.restart() and go to step 1.

After some day of researching, i found, that it's bad to powercycle modem every time and its better to use AT+CFUN=0 + AT+CFUN=1 commands, according to this document 
[Notes on the Operation of SIMCom M2M Modules Powered On and Down for Search Networks_V1.00.pdf](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/files/14968474/Notes.on.the.Operation.of.SIMCom.M2M.Modules.Powered.On.and.Down.for.Search.Networks_V1.00.pdf)

So, what i am doing wrong, or maybe its a hardware error ?

closed 2024-05-30 by github-actions[bot]