Xinyuan-LilyGO/LilyGo-Modem-Series #88: Precision regarding products

guilbaudl, 2024-03-03:
Hello,
According to the table on the repo, it seems that T-A7670G is the most capable device you sell, especially regarding GPS.
[image]

However, in this issue (https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/57) you are saying that it doesn't support GPS. 

What do I need to buy to support GPS out of the box? Do i need to buy an extra module?

lewisxhe, 2024-03-05:
See https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/56#issuecomment-1672628977

closed 2024-04-19 by github-actions[bot]

melkras, 2025-01-18:
What a dissapointment, I bought this product because of `GPS_Builtin` ✅.