Xinyuan-LilyGO/LilyGo-Modem-Series #89: Pinmaps A7670 v1.0

leakypixel, 2024-03-07:
Hi, I have the GPS version of T-SIM-A7670E and have attempted to set this up on Tasmota following the diagram [here](https://cdn.shopify.com/s/files/1/0617/7190/7253/files/T-A7670G_600x600.jpg?v=1676363553), but this doesn't seem correct. I have also tried modifying a [similar template](https://templates.blakadder.com/lilygo_T-A7608E-H.html) and also no luck.

Do you have a Tasmota template for this, or could you please confirm the pinmap? Any help would be greatly appreciated!

Thanks

lewisxhe, 2024-03-10:
View the Pinmap defined by the sketch
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/cbbc08b0134da2141818e7724b838a423c0613d0/examples/ATdebug/utilities.h#L40

leakypixel, 2024-03-11:
Great, thanks!

closed 2024-03-11 by leakypixel