Xinyuan-LilyGO/LilyGo-Modem-Series #92: Battery buck convertor possibly failed. High pitch noise and modem not starting

deeja, 2024-03-25:
I have 2 boards (A7670SA and A7670G) that are making a high pitch noise when running off battery. 
When running off battery, 
1. there is a noise,
2. the modem doesn't power on, but the ESP module seems to work.

When running off USB, 
1. no noise
2. modem works

These boards have been running for a few months off solar.
Is there anything I can do to fix this?

lewisxhe, 2024-04-06:
What voltage do you mean when the battery is exhausted?
The circuit uses a DCDC boost converter. The whistling should come from the inductor. It should be that the battery voltage is no longer enough for the booster to work.

deeja, 2024-05-10:
Thanks for updating. The project I'm working on is somewhat on hold. I will do some tests and see what voltage they are cutting out at. I think it might be related to the antenna too, but I will need to check when I have some time

closed 2024-05-10 by deeja