Xinyuan-LilyGO/LilyGo-Modem-Series #93: LORA and SD card conflict

kadrimanaj, 2024-03-25:
I'm using SPIExample.ino but when I get in the same time and SD card and LORA  one of them doesn't work.

if I put an SD card 

23:17:02.807 -> SD Card Type: SDHC
23:17:02.807 -> SD Card Size: 3840MB
23:17:02.807 -> Starting LoRa failed!

if I remove the SD card 

23:17:17.306 -> Card Mount Failed
23:17:17.306 -> Sending packet: 0
23:17:18.330 -> Sending packet: 1
23:17:19.336 -> Sending packet: 2
23:17:20.332 -> Sending packet: 3
23:17:21.355 -> Sending packet: 4
23:17:22.341 -> Sending packet: 5

I see that pin 5 is in Lora and the SD card is this pin conflict?
ore the library has a problem or example code?

lewisxhe, 2024-03-27:
You need to confirm whether the defined LoRa SPI pin conflicts with the SD card. You need to change the Pin. As mentioned in the example, you need to confirm whether the pin is correct before use.

kadrimanaj, 2024-03-27:
SPIClass RadioSPI(VSPI);

// Setup function for initializing hardware and communication interfaces
void setup()
{

  SerialMon.begin(UART_BAUD); // Start serial monitor communication
  while (!SerialMon);

     /* SD CARD Setup */
    if (SD_Card()) {
        Serial.println("Using SD Card.");
        sd_card = true;
    } else {
        Serial.println("Without SD card");
        sd_card = false;
    }

    /* Lora Setup */
    SerialMon.println("LoRa Receiver");

    
    RadioSPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI);
    LoRa.setSPI(RadioSPI);  

    LoRa.setPins(LORA_SS, LORA_RST, LORA_DI0);
    if (!LoRa.begin(FREQUENCY)) {
      SerialMon.println("Starting LoRa failed!");
      while (1);
    }

    LoRa.setSignalBandwidth(BANDWIDTH);
    LoRa.setSpreadingFactor(SPREADING_FACTOR);
    LoRa.setCodingRate4(CODING_RATE);
    LoRa.setTxPower(OUTPUT_POWER);
    LoRa.setPreambleLength(PREAMBLE_LENGTH);
    LoRa.setGain(GAIN);
    SerialMon.println("Lora ready.");

   }
   
   Fixed

closed 2024-05-11 by github-actions[bot]