Xinyuan-LilyGO/LilyGo-Modem-Series #97: Unable to see Serial device on OSX  / Macbook Sonoma

pacmac, 2024-04-03:
Hi;

Just received a couple of these, sorry to say but there seems to be a huge lack of documentation for this product.

I downloaded and installed the nCH341SER_MAC drivers, however I am sceptical that they actually installed, and besides I have so many other USB Serial Adapters and ESP32 development boards and none of them have ever required me to download a specia driver.

It has 2 USB ports, one USBC, and the other Micro USB - why what are the different USB ports for ?

If I plug into the USB C, then the LEDs on the board do not come on and no serial device is created under /dev lsusb shows no device connected.

If I plug into Micro USB, then the LEDs on the board come on, however no Serial device is created under /dev and lsusb shows:

```
lsusb
Bus 001 Device 001: ID 1e0e:9011 1e0e Mobile Composite Device Bus  Serial: 200806006809080000
Bus 000 Device 000: ID 1e0e:9011 1e0e USB 3.1 Bus  Serial: 200806006809080000
```

I have tried holding GP0 button and pressing Reset/Enable but that makes no difference.

So thus far I have 2 boards and are unable to use them :-(

lewisxhe, 2024-04-05:
Open the Serial Monitor, press the reset button, is there any startup message?

pacmac, 2024-04-05:
Which USB to connect to USB-C or Micro-USB ?

lewisxhe, 2024-04-05:
USB C

pacmac, 2024-04-06:
Hello, thanks for helping.

Now when I plug it into USB-C the serial port is recognised.

I am using Aruino, is there any sketch already installed on the device ?

Also, what is the Micro-USB Connector for ?

Thanks

pacmac, 2024-04-06:
Hi;

I can get the LILYGO AT Modem interraction on USB-C, I can get it to enter bootloading mode by pressing & Holding GP0 and then RST, but when I attempt to either upload a sketch or burn the bootloader I get:

```
Sketch uses 260685 bytes (19%) of program storage space. Maximum is 1310720 bytes.
Global variables use 21344 bytes (6%) of dynamic memory, leaving 306336 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port /dev/cu.usbserial-56750109731
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a0:b7:65:fe:c3:a8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.

A fatal error occurred: Unable to verify flash chip connection (Invalid head of packet (0xE0): Possible serial noise or corruption.).
Failed uploading: uploading error: exit status 2
```

I am using very good quality USB C cable and can program many other devices without any problem so this is not related to my cables / hardware.

pacmac, 2024-04-06:
OK, I slowed down the USB speed from 921600 to 460800 and I am able to upload sketch.

What sketch was running on the default board ?

lewisxhe, 2024-04-06:
Default is [ATDebug](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/ATdebug)

pacmac, 2024-04-07:
Hello again;

I seem to be unable to power off or prevent the modem from powering on:

```
#include <driver/gpio.h>

// The modem boot pin needs to follow the startup sequence.
#define BOARD_PWRKEY_PIN                    (4)
#define BOARD_POWERON_PIN                   (12)

void setup() {
  Serial.begin(115200);
  while(!Serial){delay(100);}
  delay(1000);
  Serial.println("Started");

  pinMode(BOARD_POWERON_PIN, OUTPUT);
  digitalWrite(BOARD_POWERON_PIN, HIGH);

}

void loop() {
  Serial.println("Loop");
  delay(1000);
}
```

I have also tried:

```
digitalWrite(BOARD_POWERON_PIN, LOW);
```

But it makes no difference, the modem always powers on.

pacmac, 2024-04-07:
I know I can use AT command to power it off, but I dont want it to come on at all until I request it to.

pacmac, 2024-04-08:
OK, seems that when powered by USB, the modem always comes on. that makes development more tricky as I cannot make it behave like it would if deployed.

But now that I understand that its OK.

closed 2024-04-08 by pacmac