Xinyuan-LilyGO/LilyGo-Modem-Series #100: the modem does not turn on when powered by the battery

Aguilaaar, 2024-04-12:
I have the board with the A7608SA-H modem, and everything works very well when I'm using power via the USB-C cable. But when I switch to the battery, the modem does not turn on, the ESP works normally, runs the code, sends commands to the modem, but it does not respond.
I have two identical boards and both have the same problem, with my code or with examples here.
Do I need to do anything different to use with batteries?

lewisxhe, 2024-04-13:
Which example to use

Aguilaaar, 2024-04-13:
I tried many of these that are here https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples
I believe that everyone who uses a GPRS connection

Aguilaaar, 2024-04-13:
I just found my mistake!
I studied the schematic files and saw that I need to keep ESP pin 12 high, as shown in the T-A7608X-V1.0.pdf file.
I hadn't found this information anywhere else.

lewisxhe, 2024-04-15:
POWERON MODEM is included in every example. Didn’t you see it?

closed 2024-05-29 by github-actions[bot]