Xinyuan-LilyGO/LilyGo-Modem-Series #101: A7670x and Wireguard

jichef, 2024-04-18:
Hello everyone. I am working with several Lilygo boards, tsim7000 and A7670E. I'm using code to make them work with the Traccar platform. Both work perfectly, I can send data to my own server successfully. However, to secure connections I want to use a Wireguard VPN according to the libraries that can be found. If I use a wifi interface, the connection is correct. But if I use a GSM connection it always gives an error.

Is there any documented problem?

rolsch, 2024-05-09:
> I'm using code to make them work with the Traccar platform.

I work on the same project for my traccar platform and vehicle tracking.
At the begin, i don't use wireguard but is in plan.

Can you share the code?

jichef, 2024-05-09:
Of course. But I am still working with the tsim7000. If you want to use the a7670 you need to use an specific libraries from here. 

The Traccar code I am using you can take from here https://github.com/onlinegill/LILYGO-TTGO-T-SIM7000G-ESP32-Traccar-GPS-tracker

you need to update pins. I think they are differents.

If in the end you manage to make it work through a VPN, I don't care which one, please let me know.

I have tried with a reverse proxy (NGINX), but I need to add a Basic authentication working with arduino. I'm also stuck there.

Good luck!

closed 2024-06-24 by github-actions[bot]