Xinyuan-LilyGO/LilyGo-Modem-Series #104: Will join network only very infrequently

ChadMatsalla, 2024-05-04:
Greetings,

My module won't join my LTE network. It **did**, and now it will not.

I'm using examples/MqttClient and on at least two occasions it joined the LTE network and sent an MQTT message. I have IOT sims from my local provider but I'm currently using the sim from my phone. The sim works in my phone, of course.

The rest of the time, modem.waitForNetwork() never succeeds.

networkMode() == 2

It just won't join the network:
```
AT+CEREG?
+CEREG: 0,0

OK
```

Can you help?

I've driven around trying different towers to no avail. I'm powering the device from a Macbook Pro and I've tried a very strong USB-C PD power supply that I use for my Raspberry Pis. Power should not be an issue.

My only theory is that the antenna is somehow not functioning.

ChadMatsalla, 2024-05-06:
```ATI
Manufacturer: INCORPORATED
Model: A7670G-LLSE
Revision: A7670M7_V1.11.1
IMEI: 867284061775948
+GCAP: +CGSM,+FCLASS,+DS
```

ChadMatsalla, 2024-05-06:
```
AT&V
&C: 1; &D: 0; &F: 0; &W: 0; E: 1; Q: 0; V: 1; X: 1; Z: 0; S0: 0; S2: 43; S3: 13; S4: 10; S5: 8; S6: 2; S7: 1; S8: 2; S9: 6; S10: 7; S11: 63; S30: 10; +FCLASS: 0; +IFC: 0,0; +CSCS: IRA; +CREG: 0; +CGREG: 0; +CEREG: 0; +CGDCONT: (1,"IP","YourAPN"); +CGDSCONT: ; +CGACT: (1,0); +CGEQMIN: (1,4,0,0,0,0,2,0,"0E0","0E0",3,0,0,0,0); +CGQMIN: (1,3,4,5,1,1); +CGEREP: +CSDH: 0; +CPMS: "SM",0,20,"SM",0,20,"SM",0,20;

OK
```
I realize that the apn is not configured but the point is that it never gets to the apn stage. It won't register.

Today it registered once and showed signal strength of 20

ChadMatsalla, 2024-05-06:
My provider @SaskTel in Saskatchewan, Canada will not permit registration without a correct APN. I'm not quite sure if this is normal because when I used a SIM from Bell Canada it permitted registration without a correct apn.

@lewisxhe Would you be willing to comment about whether some carriers permit registration with an incorrect APN? On Bell, I saw this:
```
AT+CEREG?
+CEREG: 0,3

OK
```
3 is denied. On SaskTel, I saw nothing but 0,0

ChadMatsalla, 2024-05-06:
I'm going to leave this here in the hopes of helping someone else some day.

ChadMatsalla, 2024-05-16:
I'm going to add one last thing for someone in the future. If you are seeing what I saw (+CEREG: 0,0) when you think it should work, open the ATDebug sketch and give this command
```
AT+CGDCONT=1,"IP","your.apn.here"
```
The root of the issue is that on a sketch like MqttClient, the correct apn sometimes needs to be set first in the modem for it to successfully register. If it doesn't register, the sketch never gets to the point where it registers the correct APN. Chicken, meet egg.

lewisxhe, 2024-05-17:
Yes, if the network is rejected, it may be an APN problem. Sorry, I did not add instructions in the example because not all operators need to configure APN. In my country, APN is issued through If it needs to be set manually, I will add comments to the example

ChadMatsalla, 2024-05-17:
@lewisxhe That looks good. I like setting the apn early with ```AT+CGDCONT=1,"IP","apn.operator.com"```

closed 2024-07-01 by github-actions[bot]