Xinyuan-LilyGO/LilyGo-Modem-Series #105: Deepsleep problem

PontiuspilatusNL, 2024-05-14:
I have tried five different boards. Version 1.2 and 1.2. 7670E, all the same problem; Works perfect when Board is always on. With deepsleep every board has the same problem; sometimes it runs for a few days, sometimes even three weeks, but most of the time after a few days board is hanging.
Doesn't matter running on battery or powersupply, same problem. looks like modem is not starting up properly. Even with the new sketch to turn of modem in deepsleep still the same problem.
Anybody any idea?

lewisxhe, 2024-05-17:
After running for several days, the modem cannot be woken up using battery or USB?

PontiuspilatusNL, 2024-06-07:
Excactly, Modem-Led is blinking, like it is connected, but program stops. Nothing goes, untill i reset or power off the board.

barrodoy, 2024-06-12:
I have the same issue. I couldn't find a solution yet.

Rikests, 2024-06-22:
This sounds exactly like the problem I was having with my A7670E LilyGO boards, they would run for months before hanging, sometimes a few days.
You should try to read the serial output through UART pins of the board, without using USB cable and see if the ESP32 chip is getting brownouts or restarts when the modem is starting up after **LONG** (more than 1hour) deep sleep periods. When you use USB to monitor the serial connection the power to the modem is delivered in a different manner, this is why this only happens on battery power.
I use CP2102 based USB to UART converter to monitor the serial through Boards UART pins.

HarrisonPace, 2024-06-26:
Is this happening just on the A7670E variant or the A7670G modem as well?

br4593, 2024-07-21:
It happens to me with the A7670G with GPS variant.

closed 2024-09-04 by github-actions[bot]