Xinyuan-LilyGO/LilyGo-Modem-Series #106: Page link to T-SIM767XG-S3 don't works and SIM7670G = SIM 7672G?

rolsch, 2024-05-16:
See table at point 1 and so on.

SIM7670G = SIM 7672G?

I see in some *.ino file examples the comment that sim7670G has NO GPS - i think this is not correct.

lewisxhe, 2024-05-17:
Please note that SIM7670G is the renamed name of SIM7672G.
Another version is A7670G, they are different platforms
SIM7670G uses the Qualcomm platform, A7670x uses the Asrmicro platform.

[image]

Regarding sending text messages, you need to be on a specific network, and the network needs to support the text message function before it can be used. This is not available in my country.

Can you tell me what network you are using and which country?

rolsch, 2024-05-17:
Hi Lewis.

Thanks for the info.

I user this network:

Germany
Telefonica (ehemals E-Plus) Deutschland: | MCC:262 (Deutschland) / MNC:03 (Telefonica, E-Plus) | Operator code: 26203

at+cops=?
+COPS: (2,"","","26203",7),(3,"","","26201",7),,(0,1,2,3,4),(0,1,2)
OK
at+cops?
+COPS: 0,2,"26203",7
OK

I can send and receive SMS with this board.

Now i test the voice features, but i need an mic/speaker assembled.
- any hints for the specs?

rolsch, 2024-05-18:
Hi Lewis,
i ask the SIMcom Support for aktual Firmware.

The respond:

_I assume you have purchased a board including the module from a Chinese web-shop right ??
Please note that the SIM7670G-MNGV are not supported in Europe and will not enjoy the certifications and for FW updates you need to contact the source from where the board was purchased.
The correct version for outside China is the SIM7672G-MNGV.
Basically the same module as the SIM7670G-MNGV._

So i see here an diffenence between SIM7670G and SIM7672G.

closed 2024-07-02 by github-actions[bot]